hamishcoleman/thinkpad-ec #1: Port battery validation patch to other machines

haarp, 2016-04-26:
Hello and thank you very very much for this project! I'm going to upgrade to a W530 just for this reason :)

However for the far future I see problems with battery availability from Lenovo. You have Zmatt's patch for the X230 in your repo, but only for that machine. Cold the same be ported to other machines? Or would that involve different or hard to find offsets?

Thanks!

hamishcoleman, 2016-04-26:
I'm glad that others are finding it useful.  Do let me know if you can
think of ways that I should make things simpler to install - I believe
one of the biggest problems is making it easily accessable to other.
My current best plan is to make the patch apply to the iso image itself,
hopefully bypassing all the questions about "where do I get dosflash.exe
from" and "what are the exact commandline options".  Which should work
for you - since (IIRC) the w530 actually has a CDROM - but I still need
to test and confirm all that with a usb cdrom drive on my x230...

I wouldnt borrow trouble - consider that by the time you need a new
battery, the clone batteries might have cracked the authcode and will
have a chip that authenticates OK.

I dont actually have all of zmatt's patch - and I had to port the part
that he did publish as he was using a slightly different firmware version
for his work.  So, I can say that there is probably enough context to
port it to other hardware, but that there would still be more work to
do at that point.

haarp, 2016-04-26:
Hey,

I actually do have a suggestion that would make things easier. There could be more information (in the README) to help new users orient themselves. Such as

- What is the point of all this to begin with? e.g. "Patch the EC to make older generations keyboards work, battery hack, etc"
- What works, what doesn't? e.g. "All keys except Fn+F3 are functional"
- Which hardware mods are necessary? e.g. Isolating keyboard pins is (not?) necessary (as mentioned [here](http://forum.thinkpads.com/viewtopic.php?f=69&t=120776), nubs will have to be cut off from the keyboard, etc.

As for the battery hack, you certainly have a point. However I still think that uncrippling all **30 would be worthwhile, even just to allow usage of **20 generation and current third-party batteries. This is at your discretion of course, I'm sure you have better things to do than hunt down the battery check on every EC image out there :)

I intend to grab a hardware programmer anyway to be able to remove the Wifi whitelist from the BIOS of the W530, so I should be able to fearlessly test EC images too.

hamishcoleman, 2016-04-28:
For a start, I dont actually have a battery that doesnt validate, so I cannot easily test any patches.

Also, you should know that the hardware programmer required for the EC is a JTAG programmer, and as far as I know, you would need to solder to a bunch of test points..  So, you might not want to be too fearless :-P

haarp, 2016-04-29:
Well, not quite. You can use clamps like these: http://i.imgur.com/jrN6biG.jpg

The problem appears to be that the 3.3V line is in parallel with all kinds of peripherals and most programmers cannot deliver enough current for that. But that can be boosted.

hamishcoleman, 2016-05-06:
I have ported the battery validation patches to all the other firmware patchsets.  Since I am totally unable to test it, I have left it disabled by default...

closed 2016-05-06 by hamishcoleman

haarp, 2016-05-06:
Oh, sweet, thanks a lot! I shall test it once I have a W530 and programmer :)