hamishcoleman/thinkpad-ec #3: Broken filename during flash

haarp, 2016-05-28:
Hey,

I just created a W530 image according to the README (tried both .iso and .img), but flashing fails.

"This disc will flash the file \"

    dosflash /sd /ipf ec /file \

That backslash can't be right.

hamishcoleman, 2016-05-28:
Yup, that's not right.

It looks like something has gone wrong with one of the mtools commands.
I've just done a test build and am not seing the same thing, so this
might due to the version of mtools.

During your build, you should see the following commands:

    + dd status=none conv=notrunc bs=5242880 seek=1 of=w530.G4HT39WW.s01D5200.FL2
    sed -e "s%__DIR%`mdir -/ -b -i g5uj28us.iso.orig@@71680 |grep FL2 |cut -d/ -f3`%; s%__FL2%`mdir -/ -b -i g5uj28us.iso.orig@@71680 |grep FL2 |cut -d/ -f4`%; s%__DESC%`cat g5uj28us.iso.orig.desc`%" autoexec.bat.template >g5uj28us.iso.bat.tmp
    mv g5uj28us.iso.bat.tmp g5uj28us.iso.bat
    ./slice.insert w530.G4HT39WW.s01D5200.FL2.slice w530.G4HT39WW.s01D5200.FL2 g5uj28us.iso && MTOOLS_SKIP_CHECK=1 mcopy -o -i g5uj28us.iso@@71680 g5uj28us.iso.bat ::AUTOEXEC.BAT

The important one here is the "sed" command. Do you see any error messages
after that command.

Can you also check what version "mtools --version" reports (On my current
test system, I have "mtools (GNU mtools) 4.0.17"

Also, the exact command that is not working will be:

    mdir -/ -b -i g5uj28us.iso.orig@@71680 |grep FL2

and if you run that manually, you might see some other error messages
(It should should get the result "::/FLASH/G5ETA5WW/$01D5200.FL2" if it
is working correctly)

haarp, 2016-05-28:
Weird, it does not appear to run sed at all on my end.

    $ make patched.w530.img
    for i in *.slice; do read SLICEE other <$i; echo `basename $SLICEE .orig`: $i `basename $i .slice`; done >.d/slice.insert.deps.tmp
    mv .d/slice.insert.deps.tmp .d/slice.insert.deps
    for i in *.slice; do read SLICEE other <$i; echo $i: $SLICEE; done >.d/slice.extract.deps.tmp
    mv .d/slice.extract.deps.tmp .d/slice.extract.deps
    ./slice.insert w530.G4HT39WW.s01D5200.FL2.slice w530.G4HT39WW.s01D5200.FL2 g5uj28us.iso && MTOOLS_SKIP_CHECK=1 mcopy -o -i g5uj28us.iso@@71680 g5uj28us.iso.bat ::AUTOEXEC.BAT
    + cp --reflink=auto g5uj28us.iso.orig g5uj28us.iso
    + dd status=none if=w530.G4HT39WW.s01D5200.FL2 bs=12718080 count=1
    + dd status=none conv=notrunc bs=3628544 seek=1 of=g5uj28us.iso
    mv g5uj28us.iso patched.w530.iso
    ./geteltorito -o patched.w530.img patched.w530.iso
    Booting catalog starts at sector: 20 
    Manufacturer of CD: NERO BURNING ROM
    Image architecture: x86
    Boot media type is: harddisk
    El Torito image starts at sector 27 and has 65536 sector(s) of 512 Bytes
    
    Image has been written to file "patched.w530.img".
    ./hexpatch.pl patched.w530.img fix-hdd-image.patch
    Attempting to patch patched.w530.img
    Applying fix-hdd-image.patch fix partition table
    
    $ mtools --version
    mtools (GNU mtools) 4.0.15
    configured with the following options: enable-xdf disable-vold disable-new-vold disable-debug enable-raw-term

hamishcoleman, 2016-05-28:
It didnt re-run the sed command because it had cached the previous result.
You can get it to re-run the sed command by removing the cached output
file - in this caes "g5uj28us.iso.bat"

haarp, 2016-05-28:
Since *make clean* didn't exist, I used *git clean -fd* instead, but it looks like it didn't remove all caches. I removed that file and tried again:

    sed -e "s%__DIR%`mdir -/ -b -i g5uj28us.iso.orig@@71680 |grep FL2 |cut -d/ -f3`%; s%__FL2%`mdir -/ -b -i g5uj28us.iso.orig@@71680 |grep FL2 |cut -d/ -f4`%; s%__DESC%`cat g5uj28us.iso.orig.desc`%" autoexec.bat.template >g5uj28us.iso.bat.tmp
    Total number of sectors (64195) not a multiple of sectors per track (32)!
    Add mtools_skip_check=1 to your .mtoolsrc file to skip this test
    Total number of sectors (64195) not a multiple of sectors per track (32)!
    Add mtools_skip_check=1 to your .mtoolsrc file to skip this test
    mv g5uj28us.iso.bat.tmp g5uj28us.iso.bat

hamishcoleman, 2016-05-28:
I was just adding a "clean" target while you were doing that - I hadnt
needed one until now :-S

That error message clears it up for me - I forgot to add an option that
that slightly older mtools needs.

I have just pushed an update to the repo, so try updating your repo and
re-running your build

haarp, 2016-05-28:
echo This disc will flash the file G5ETA5WW\$01D5200.FL2

Yeah, this looks correct. I already flashed it manually prior to your fix and the new ec worked perfectly too.

Thanks! :)

closed 2016-05-28 by haarp