hamishcoleman/thinkpad-ec #5: T430s

jacoblharris, 2016-06-05:
I bought a w500 FRU 45N2211 keyboard to replace my T430s keyboard.  I haven't had any luck remapping the keys via the EC flash. I have followed the steps in the wiki and the readme on the git repo but nothing changes. I have tried on a flash drive and on a CD I get a successful flash message but the key mapping doesn't change. It's almost like I'm just reflashing the stock firmware over the original and not the modified one.  Any advice?

hamishcoleman, 2016-06-06:
If you are not getting any errors from the flashing, but the keys are not remapped, it sounds like the firmware build did not work properly.

Can you pull the latest updates from the repo and run a "make clean" followed by whatever you were building (I suspect "make patched.t430s.img") - then paste the text output of the build into the ticket.

It also might be useful if you can paste the output from "sudo dmidecode -t bios"

jacoblharris, 2016-06-06:
I copied my entire terminal session, maybe you will catch something I did wrong.

[BIOS T430s.txt](https://github.com/hamishcoleman/thinkpad-ec/files/299700/BIOS.T430s.txt)
[Terminal T430s.txt](https://github.com/hamishcoleman/thinkpad-ec/files/299699/Terminal.T430s.txt)

hamishcoleman, 2016-06-06:
nope, I can see nothing wrong in your build - and you are getting the same sha1sum that I get when I do a test build - so that is good.

Looking at the dmidecode data, it might be that your BIOS is too old to allow the flashing of this EC version (the one you are running is 14 versions older than the latest)

So, it might be worth trying again if you can upgrade your BIOS with the latest unpatched version first.

jacoblharris, 2016-06-06:
That did it, I updated to the latest BIOS and it rebooted, I got a message on the POST screen saying flashing embedded controller.  Once it was done updating the BIOS I ran the modded EC firmware and it rebooted with a "flashing embedded controller" message in the bottom left, I had never seen that in the past.  Thank you very much for your help and time!

hamishcoleman, 2016-06-06:
Fantastic!  I'll add that hint to the readme.  There is some lenovo documentation that says that you can run a newer EC than your BIOS - but clearly that does not always apply...

closed 2016-06-06 by hamishcoleman