hamishcoleman/thinkpad-ec #6: 7-Row keyboard trackpoint non-functional

Mikejmoffitt, 2016-06-29:
Following the recommended procedure, where pins 25, 27, and 29 are covered on the X220 keyboard's internal connector, the trackpoint buttons do not work with the X220 keyboard on an X230.

Mikejmoffitt, 2016-06-29:
In addition, attempting to revert to the original keyboard is difficult, as I can not seem to re-flash the embedded controller with the stock firmware.

closed 2016-06-29 by Mikejmoffitt

hamishcoleman, 2016-06-30:
I assume you got it all working?  was there anything we could add to the instructions that would have helped you to succeed faster?

Mikejmoffitt, 2016-06-30:
I fixed the trackpoint buttons, but closed the issue as that's likely more hardware related, given the success others have had. Regarding reverting, it would be nice to have a make target to create a bootable image that just writes the original EC firmware, but I managed to do so myself with a little hack so the X230 is at least usable.

I figured further discussion and troubleshooting would be most appropriate in the thinkpads.com thread, since I doubt this is really a software issue.

damex, 2016-07-16:
the problem is - you done tape method for pins wrong.
i have had same issue and many others trying to tape pins unless i got some screen protection film and cutted it to tape pins.

ferrao, 2019-03-01:
@Mikejmoffitt I have the exact same issue  with trackpoint buttons now working after the procedure.
Can you please share what you did to fix it? Thanks!