hamishcoleman/thinkpad-ec #7: Patching image fails, T430

pcbmaster, 2016-07-13:
`Image has been written to file "patched.t430.img".
./hexpatch.pl patched.t430.img fix-hdd-image.patch
Attempting to patch patched.t430.img
Address 448 mismatched data
The binaryfile does not match the context bytes from fix-hdd-image.patch
make: *** [Makefile:210: patched.t430.img] Error 1
`

I don't know why it's failing.  Running Arch.

pcbmaster, 2016-07-13:
Fails at the same point under Debian 8 as well.

hamishcoleman, 2016-07-14:
It is failing at one of the very last steps.  Your "patched.t430.img" is actually ready to use, but could not be patched to allow testing with qemu.

I have just run a quick test here and can see that the original Lenovo t430 ISO image is constructed slightly differently to the other ISO images - so the "fix-hdd-image.patch" does not apply properly.  I'll have a look at what is different here and work on a fix

hamishcoleman, 2016-07-14:
I have updated the repo with a fix for this - give it a try

pcbmaster, 2016-07-14:
It completes without error, but the keyboard patch doesn't work

hamishcoleman, 2016-07-14:
Can you give me any details?

pcbmaster, 2016-07-14:
Never mind, it finally flashed the controller.

closed 2016-07-14 by pcbmaster

hamishcoleman, 2016-07-14:
Was it just slow? Or was there something else that I might be able to document?

pcbmaster, 2016-07-16:
After the successful flash and it rebooted, it just booted to nothing, since I use UEFI for everything installed.  After it showed the splash and sat on a black screen for a while, I turned it off.  I then booted back up, went into the bios, changed back to UEFI mode, then saved and rebooted.  When it came back on, THEN it flashed the EC. Super odd.

On an unrelated note: how can I build with the battery patch?

hamishcoleman, 2016-07-19:
You can build with the battery patch by renaming the patch file in the appropriate subdir so that it doesnt have ".OFF" and then doing a "make clean"

Sorry for the delay - I wrote this up the other day (along with some notes in the README) and only just realised that I had not pushed it