hamishcoleman/thinkpad-ec #8: Applying other patches

haarp, 2016-07-16:
The directory structure and Makefile are very cluttered and complex. I'm trying to figure out how to apply the keyboard patch to a W530 image. is it sufficient to strip the *.OFF* from the hexpatch filename? Because I'm still getting the same hash at the end:

echo Built: f33c31485358bd314bdb5eefe20f3a894baa36ae  w530.G4HT39WW.s01D5200.FL2

haarp, 2016-07-16:
Ok, I think I figured it out. 

Running *make patched.foo* will not re-evaluate the patch files. *make clean* does not help either. You have to run *make really_clean.*, then redo *make patched.foo* for it to pick up renamed patches.

Can you confirm that this is the correct hash?

echo Built: f33c31485358bd314bdb5eefe20f3a894baa36ae  w530.G4HT39WW.s01D5200.FL2

hamishcoleman, 2016-07-16:
Yeah, the makefile is a little messy - it all kind of grew organically.  I also did not expect to support so many combinations.  I have thought about rearranging it, but as it is now pretty much at its end state there doesnt seem to be much value to be had in doing that.

I've just run the w530 build with both the keyboard and battery patch applied and do get the same hash as you:

echo Built: f33c31485358bd314bdb5eefe20f3a894baa36ae  w530.G4HT39WW.s01D5200.FL2

haarp, 2016-07-16:
Perfect, thanks!

closed 2016-07-16 by haarp