hamishcoleman/thinkpad-ec #9: latest update placed t430 info into t430s sections

cmorche, 2016-08-05:
I left comments on the code. i dont know how to use github, but i hope it can be resolved easily

hamishcoleman, 2016-08-05:
That does sound like the kind of silly mistake that I would make.  I'll go and doublecheck the right firmware versions and do an update for you

cmorche, 2016-08-05:
Thanks dude! I'm learning to code, but am noob right now.

hamishcoleman, 2016-08-05:
I've just looked at https://support.lenovo.com/au/en/olddownloads/ds029725 and it says that g7uj19us.iso is for ThinkPad T430s and T430si - which looks like what I was trying to update with that commit (though the commit message doesnt say t430s).  I have found that I duplicated a line erroneously in the makefile (which I have just fixed), but I'm unsure if that was the issue you were seeing.

Can I confirm what you were trying to do?

cmorche, 2016-08-05:
Are you sure it doesn't say g7uj18us.iso ? Lenovo made it easy to mistake them. The 19 is for the t430 and the 18 is for the t430s. I have a T430s and am trying to flash this now, but noticed that after booting into the USB it would flash but the keymaps never change (and the BIOS never goes into an EC flash mode)

Thanks so much for you work on this!!

cmorche, 2016-08-05:
Well, im not sure what I actually did... I just rebooted my laptop with the intent to try and flash my updated makefile version and then it said "updating EC" and now its working great!! 

Now to figure out how to hold this keyboard in place better....

hamishcoleman, 2016-08-05:
Yeah, Lenovo doesnt really make it straight forward to see the 'real' names of their updates, do they.

The pages I'm looking at appear to show g1uj38us.iso is for T430 - when I look at all the other iso names, I think it goes like this:

MMTTVVCC.iso

Where:
* MM is the "Model" of the laptop - so  indicating T430 or T430i or X230 (etc)
* TT is the "Type" of product - all the laptop ISOs that I'm looking at have "uj" here
* VV is the version
* CC looks like a country code (but all these ISOs are "us" here)

hamishcoleman, 2016-08-05:
Just as I was finishing that, I saw your update - sounds like good news!  Always happy to hear of someone's success :-)

I assume you are using the original t430 palmrest? - you can just file down the nubs on the keyboard to make it fit well enough that the screws will go in properly

cmorche, 2016-08-05:
Thank you for the information!

I am using the original t430s palm rest, so I have filed down the nubs. The
only issue is that I have the full HD mod, which pushes the keyboard up a
bit on the left side. Since the nub is gone it sometimes pops up a bit. I'm
just going to put a piece of double stick tape in there to hold it down.

Cheers!

hamishcoleman, 2016-08-05:
Ah, I had not seen that mod - that might make it a little more difficult..

hamishcoleman, 2016-08-05:
now, if only I knew why your update didnt work straight away - which has happened in the past, but we thought we narrowed it down to the UEFI mode vs Compatibility mode

closed 2016-08-10 by cmorche