hamishcoleman/thinkpad-ec #10: Feedback: T430, G1HT35WW battery patch

diggit, 2016-09-01:
Hi, this is awesome project!

I've flashed EC on my T430 with G1HT35WW and applied battery patch only. 55+ battery started charging. Do you need some more test or is this sufficient to consider battery patch as tested and working?

koba213, 2016-09-01:
Hey,

I have T430 with G1HT35WW (2.69 BIOS) and did the battery patch only as well, but it still is not charging the 55++ battery. It only got rid of the annoying "battery is not charging" notification at boot.

diggit, 2016-09-01:
Wow, I have pretty archaic BIOS version:
```
# dmidecode 3.0
Getting SMBIOS data from sysfs.
SMBIOS 2.7 present.

Handle 0x000E, DMI type 0, 24 bytes
BIOS Information
	Vendor: LENOVO
	Version: G1ET41WW (1.16 )
	Release Date: 05/25/2012
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 12288 kB
	---shortened---
	BIOS Revision: 1.16
	Firmware Revision: 1.13
```
But charging 55+ works great. Now I tested 55++ and it works ok too.
I have applied start and stop charging thresholds, reconnected AC power, nothing more.

70+ or 70++ batteries works ok for you?

koba213, 2016-09-01:
Yeah, I don't have a compatible battery. I was on the 2.63 BIOS before and I was able to flash G1HT34WW successfully but I upgraded BIOS for whole other reason so I'm now stuck with useless paperweight of a battery and un-downgradeable BIOS.

diggit, 2016-09-01:
> ...so I'm now stuck with useless paperweight of a battery and un-downgradeable BIOS.

I bet you can still downgrade it, it only depends on how courageous you are. [Look here.](http://www.thinkwiki.org/wiki/Custom_BIOS)

I've flashed 2.71 BIOS (EC flashing didn't occur, probably due to being up to date thanks to previous EC flashing) and battery is charging.
At first it looked like it does not, so I set start and stop limits using tpacpi-bat
```
tpacpi-bat -v -s ST 1 99
tpacpi-bat -v -s SP 1 99
```
and reattached power cord. Charging works. Again, IDK if it was necessary, but no problem afterwards. This was probably caused by *TLP* which sets some more sane limits.
Charged from 95% to 98% just for test. It stops there, but who cares...

If you don't want to wait for percentage changes to check battery progress, read battery energy:
`cat /sys/class/power_supply/BAT0/energy_now`, number should rise. `status` in same path should read `Charging`.

Successfully tested charging at T430 with EC patched with battery patch only:

| BIOS | EC | battery |
|---------|-----|------------|
| 1.16   | 1.13| 55+, 55++|
| 2.71   | 1.13| 55++|
*55+ or 55++, IMO one works -> the other one should too*

koba213, 2016-09-01:
Maybe I'm doing it wrong. I removed .OFF from the battery patch and added .OFF to other patches in the folder then built from there. Is this the way you did it too?

diggit, 2016-09-01:
Yes, just pull latest version from git and be sure to clean everything before executing, run `make really_clean`. (or you can clone whole repo again and rename patches...)

You can spot what patches are being applied somewhere in output:

```
...
Attempting to patch t430.G1HT35WW.img
Applying t430.G1HT35WW.img.d/006_battery_validate.patch -7220,14 +7220,14
...
```

koba213, 2016-09-01:
Well, I'll be damned, it IS charging! Thank you very much!

diggit, 2016-09-01:
Good to read that! Did you change anything?

@hamishcoleman can you mark battery patch for T430 as tested and working?

koba213, 2016-09-02:
I installed tpacpi-bat and ran these:
```
tpacpi-bat -v -s ST 1 99
tpacpi-bat -v -s SP 1 99
```
Unplugged the charger,plugged it back in and voila!

diggit, 2016-09-04:
@koba213  Maybe you (we) just have limits set to start charging in deeper discharge that battery was.

Anyway, you can set those limits back to some sane levels, using `tpacpi-bat` or some DE specific tool.

hamishcoleman, 2016-09-15:
@diggit I've updated the comment on the t430 battery patch as known good.

I have not played with tpacpi and battery charging settings, so I'll have to try that now.

maxgvz, 2016-11-12:
Hi everyone!

Do you guys can please share patched BIOS .ROM file for flashing T430 ?
Please, please. I beg you. 

Thank you!

maxgvz, 2016-11-12:
Hi everyone.

Thank you for this patch! It is really working! Just patched my BIOS and no more problem with the batteries validations! Thanks again and have a good one.

closed 2016-11-13 by diggit

gornostaevdev, 2017-08-07:
2.72 and the only applied is t430.g1ht35ww.img.d/006_battery_validate.patch

passed both bios&ec flashing, battery still not charging (66%, do not powering on without AC, turns off when AC unplugged).

diggit, 2017-08-10:
This is really weird. With stock (unmodified BIOS/EC), you should be able to run from older batteries and switch to AC when connected (no charging). If your battery did not work this way before flashing modified BIOS/EC FW, you have broken battery.
With patch applied, charging is possible nothing else should change. Behavior you describe sound like something is really wrong.

You should open new issue for this.

@hamishcoleman did you ever encountered such problem?

Shchekin, 2017-09-08:
Hello! It is possible to do the same for the T530?

maxgvz, 2017-09-08:
Re: [hamishcoleman/thinkpad-ec] Feedback: T430, G1HT35WW battery patch (#10)

Hello

I don't think so. Different bios.

Best regards.
--

Friday, September 8, 2017, 7:38:14 AM, you wrote:

Hello! It is possible to do the same for the T530?
—

mailto:gvozdevmax@gmail.com

hamishcoleman, 2017-09-08:
> References

Are you asking if it is possible to patch the t530 to charge non-genuine
batteries?

Yes, that is possible - the t530 is a supported system and the battery
patch is included in this repo.  You should be able to follow the
README instructions, you should be able to build a patch.

Blimthorpe, 2021-06-03:
I too patched my t430.  Battery only in Mint 20.  It removed the battery notice, but the battery in NOT charging.  It just says "26% available (plugged in).  I'm using the battery in Windows 10.  It flashed to 2.81.

Anyone have a suggestion ?

Thx

Blimthorpe, 2021-06-03:
It was a Windows problem.  I deleted the battery acpi drivers in Windows.  Took out battery, unplugged charger.   Now it is charging.  Problem solved.  Great patch.  Thanks for all the hard work.