hamishcoleman/thinkpad-ec #11: Modding EC with latest Bios

Mediazept, 2016-12-13:
Lenovo updated most of their BIOS for most of their business models, because of a security issue. How do I adapt the build process to use the latest BIOS version (g5uj29us for my W530)? Or does this patch just flash the EC and leaves the rest as is?
I was reading the thread on forum.thinkpads.com and was trying to find information at github, but to no avail.

diggit, 2016-12-13:
Look into [README](https://github.com/hamishcoleman/thinkpad-ec#readme):
> This software expects to be run under Linux.  For best results, ensure you
have updated your BIOS to a recent version before starting.  If there is too
large a difference between the BIOS and EC versions then the flash process
will not complete.

So yes, these patches affects only EC FW. BIOS is not updated at all. That is up to user.

Update of EC is not bound to BIOS update, but it is good to keep them up to date. (I updated EC and It worked with 4 years old BIOS - so not 100% necessary...)

AFAIK EC FW is not shipped with each BIOS update, so theme might not be any change in EC.

m-kozlowski, 2016-12-13:
From what i've seen recent Lenovo updates affects only bios version. EC version stays the same.
Therefore boot images created with this repo will work for current Lenovo releases.

hamishcoleman, 2016-12-15:
I've just had a quick look at two of the BIOS updates - they have not changed the EC version, so confirming what others have already said - everything is still fine to use images built from this repo.

(eg: looking at the x230 g2uj24us.iso, they have not changed the EC version for 7 releases)

Mediazept, 2016-12-15:
Thanks @all for your help. I will give it a try.

Mediazept, 2016-12-20:
I can confirm that everything is working as expected. 
Thanks for providing us with such an easy solution.

closed 2016-12-20 by Mediazept