hamishcoleman/thinkpad-ec #12: Windows compatibility

Akrome, 2016-12-15:
Hi there.

Thanks for your work. I don't expect you to port it to Windows by any means, but I wanted to ask you whether this could work by running Linux VM in a Windows container, and if so if this should be added to the instructions.

Thanks

nitrocaster, 2016-12-15:
I ported [mec-tools to Windows](https://github.com/nitrocaster/mec-tools) and used it to patch my EC. A lot of work has been done since that moment, but it works and you can use it.

Akrome, 2016-12-15:
Thank you, I'll try. Just to make sure (this isn't exactly my alley), I would still use the images that I find in this repo just push them with mec-tools, right?

hamishcoleman, 2016-12-15:
Akrome, this all works the same in a Linux VM - nothing in the instructions (until the booting from your USB stick) requires physical hardware.

You can also follow nitrocaster's process - however, you it looks like you will find this needs a bunch of manual steps for you to get a final result.

Also, there are no actual firmware images in this repo (it automatically downloads the originals that needs) just patches that it applies to the original binaries.

hamishcoleman, 2018-02-05:
Unfortunately, Windows support is not a priority (though, patches welcome)

closed 2018-02-05 by hamishcoleman