hamishcoleman/thinkpad-ec #14: Does FN+Space still control the light on x230 with the EC firmware flashed?

TheChuckster, 2017-01-24:
I have installed the x220 keyboard into my x230 but I have not flashed the EC firmware mod yet. Right now, the FN+Space controls my Thinklight, which I really do like. My question then is, if I flash the EC firmware, will my FN+Space still work afterwards and control my Thinklight?

hamishcoleman, 2017-01-25:
With the default patch, the Thinklight is controlled by the Fn+PgUp.

However, this is possible to adjust in the patch - the thinklight control is patched in the "005_fn_key_swap.patch" file.  There is a line that has "c6 55" in it - where "c6" indicates the thinklight and "55" refers to the PgUp - changing the "55" to "3d" will make it use the Space key instead

closed 2017-02-06 by TheChuckster

chuckhacker, 2017-04-17:
Thanks @hamishcoleman ! That's great!