hamishcoleman/thinkpad-ec #15: T430 revert keyboard patch?

erftrunner, 2017-03-31:
Hello,

I patched my T430 with keyboard and battery patch in place. Works fine!
Now I want to revert the keyboard patch (leaving in place the battery patch).
With no success!

make patch_enable_battery clean
make patch_disable_keyboard clean
make patched.t430.img

"Burn" the image to a USB Stick and boot.
Updating works without noticeable errors, but the keyboard is still patched.

Am I missing something?
Current BIOS version is 2.72.

thanks
Peter

erftrunner, 2017-03-31:
Ok, all well that ends well. ;-)
Made another try and this time there was a message "flashing ..." after reboot that wasn't there last time. 
Issue solved.

closed 2017-03-31 by erftrunner

hamishcoleman, 2017-04-01:
Well, that is good news.  There have been a number of cases just like yours - you didnt happen to notice something different that you did on that reboot? (that might have made it trigger)

erftrunner, 2017-04-01:
No, unfortunately not. I am myself a Software Developer and hate unreproducable errors.I did nor changed nothing specialRegardsPeter

hamishcoleman, 2017-04-05:
Thanks for the update anyway - I'll just keep asking people to 'retry'  :-(