hamishcoleman/thinkpad-ec #19: T430 2.72 bios

SquirrelCZE, 2017-06-13:
Hi folks, I updated my BIOS to last one
(dmesg msg:
[    0.000000] DMI: LENOVO 2349GCG/2349GCG, BIOS G1ETB2WW (2.72 ) 01/31/2017
)

But found out that battery patch for it from previous version is not applycable:
(I just replcaed filename for iso of bios that is downloaded and used)

./hexpatch.pl t430.G1HT35WW.img t430.G1HT35WW.img.d/*.patch
Attempting to patch t430.G1HT35WW.img
Address 115504 mismatched data
The binaryfile does not match the context bytes from t430.G1HT35WW.img.d/006_battery_validate.patch

How to approach this?

hamishcoleman, 2017-06-15:
I just had a look at the lenovo version details on http://pcsupport.lenovo.com/hk/en/downloads/DS029253
and you are in luck - the embedded controller version has not changed - so the old patches can be applied.

Unfortunately, the way the build system was originally written means that you need to work out a "magic offset number" to go in the t430.G1HT35WW.s01D2000.FL2.slice file.

I have done a quick update to this version, so if you update your repo you should be good to go.

ypid, 2017-06-15:
First of all, great work guys :+1:

> Unfortunately, the way the build system was originally written means that you need to work out a "magic offset number"

How do you currently do this? I guess opening a hex editor and searching for the copyright which seems to be at a fixed offset at the beginning of the ec firmware image or binwalk?

hamishcoleman, 2017-06-16:
The *.FL2.slice files were a shortcut to avoid needing to understand the filesystem inside the ISO file and the magic numbers they contain are the byte-offset of the start of the .FL2 file and length of that file.  This is then used to "copy" the file both out of the ISO image and back into it after patching.

I calculate these numbers by manually copying the .FL2 file out with mtools+mcopy and then searching in the ISO image for the same hex bytes - thus discovering the offset.  So, the *.FL2.slice files could just be replaced with an automatic use of mcopy

closed 2017-10-09 by hamishcoleman

t430git, 2020-03-07:
can i downgrade t430 ec v1.14...
bio v2.81
what can i do to apply battery patch...