hamishcoleman/thinkpad-ec #27: Upgrade from SHA1 to SHA2-256 for checksums

ypid, 2017-06-16:
This would provide cryptographically strong checksums. Alongside with OpenPGP signatures on commits and tags (my PRs are signed) this would improve the overall authenticity of files downloaded from remote servers. (OK, git still relays on SHA1 but that will hopefully get fixed one day).

Would you be ok with that? Can I do this?

hamishcoleman, 2017-06-16:
I did think about doing this when I set up the old .sha1 files and my main two thoughts were firstly that the sha-256 were starting to get unwieldy large and secondly that the main point of the checksums was to try to make it hard for people to shoot themselves in the foot - not a cryptographic check.

I've just gone and looked at an example lenovo information page (http://support.lenovo.com/hk/en/downloads/ds029188) and they do show sha-256 values for their downloads (and importantly, that one matched locally when I calculated it - proving that they haven't done something strange)

Perhaps if the Descriptions.txt was changed into a multi-line format then the size of the checksum would be less painful - I have been thinking about making that file a mult-line format anyway, as that would open up the possibility of more structured data (ie: g2uj25us.iso.orig has checksum $a and desc $b and contains file x230.G2HT35WW.s01D3000.FL2.orig - which could be parsed to autogenerate deps and remove some of the hacky lists from the makefile)

Hrm. I guess that adds up to "patches welcome" and "dont make it harder to use"  :-P

hamishcoleman, 2017-09-05:
I have implemented support for checking sha-256 checksums against downloaded files in commit 361e37003d705a7c57c9c25f84d6305bce0b000d

Lenovo have changed to publishing their downloads without sha-1 checksums, so any new downloads added will be using the sha-256 as published by Lenovo.

Feel free to create a patch to update the other checksums to the sha256 value that Lenovo has published.

closed 2017-09-05 by hamishcoleman