hamishcoleman/thinkpad-ec #33: X230 keyboard in X220

duairc, 2017-06-26:
Hi,

So I've been happily using these patches for a while to use an X220 keyboard in my X230.

I have a friend who uses an X220, who has a problem with their keyboard, and instead of getting a new one, I thought it would be nice if I could reuse the X230 keyboard that came with my laptop (which I still have) in their X220 (which I think they would be inclined to prefer anyway).

Now I don't have a good understanding of the mechanics of how the stuff in this repository works, but is it possible that this would be a simple thing to do with what's already there?

hamishcoleman, 2017-06-27:
So, this is kind of a complicated answer.  

There is enough information available to make a patch to the x220 firmware for this, but we dont know the details of the encryption for the x220 firmware, so we cannot create a working and patched firmware.

However - and unlike installing the x220 keyboard in the x230 - with the direction you are going, there are no dead keys.  They all generate a keypress event and thus you can use normal keyboard remapping programs to re-arrange the keys.  I can tell you how to do that with Linux, but you'd need to search for details on other operating systems.

duairc, 2017-06-27:
Okay, well that might suffice then. And yeah, they use Linux. I read on ThinkWiki that I need to use "udev keymap" to map the keys — is that how you would do it?

hamishcoleman, 2017-06-28:
Well, you caught me - I dont actually know how to do it properly on a modern linux system..

I'd probably just hack something together to call loadkeys(1) in rc.local, but it looks like a modern system has some integration and you can get your keyboard working right with loadkeys and then run "sudo setupcon --save" to save the current keymap for next boot.

closed 2018-02-05 by hamishcoleman

matrushko, 2019-11-05:
hi, i have a x230 and his own keyboard. 

But I mistook something aplying battery patch and my keyboard went crazy , fn, supr, up down keys, etc. 

i dont know if i should make another usb flash with @enable@ or @disable@ keyboard patch)__
how can I go @back@ to my original function keys+

sorry I can-t find the special characters in linux.