hamishcoleman/thinkpad-ec #38: T430s BIOS 2.67

orbitinstasis, 2017-07-28:
Hello,

Some security changes are made in the later BIOS versions that the script fetches, it fetches 2.65 but the latest on the t430s is 2.67. 

Is it possible to change this in the script, and if so, how? 

I assume flashing the patched 2.65, then updating to 2.67 the via the intended method wipes the ECU's patches ?

Cheers

hamishcoleman, 2017-07-28:
Yes, it is possible to change the script - and it is a simple change because the EC version had not changed.  So, I have just made that change.

As you noticed, your BIOS update didnt wipe out the EC change - this is because there was no version change in the EC firmware, so it just skipped flashing the EC.

orbitinstasis, 2017-07-28:
Awesome, but were the patches written ? Asking because you said it skipped flashing the EC. A bunch of shortcuts on the chiclet keyboard no longer work so i'm assuming it worked.

hamishcoleman, 2017-07-29:
I meant that the official lenovo BIOS update will notice that the currently installed EC version was exactly the same - and skip performing any update to the EC.  So, if you updated your BIOS to get the security updates, you will not have overwriten any keyboard patch installed in the EC.

It sounds from your description of keyboard shortcuts that you do have the keyboard patch installed successfully (I assume you still have the original t430 keyboard in the machine?)

orbitinstasis, 2017-07-29:
Yes, still not switched keyboards. Before flashing the patch all shortcuts worked as expected. 
I'm guessing by your description that the flash shouldn't have worked? Now on the chiclet keyboard, fn+home inscreases brightness, but fn+end doesn't do anything; similarly i can't turn on the think-light led thing. I'm guessing it's because I'm still on the chiclet. 

I might pull the new changes  + recompile + flash with new bios for good measure.

orbitinstasis, 2017-07-30:
I just pulled the changes, turned off the battery and keyboard patches, compiled and flash, but the bios (despite writing the bios) didn't get to the 'flashing ecu' part, and no changes were made (battery and keyboard patches still live).

hamishcoleman, 2017-10-09:
It sounds like the flashing worked perfectly to me.  The reason why Fn+End doesnt do anything is that that keyswitch is actually no longer "End" on the old xx20 keyboards

closed 2018-02-05 by hamishcoleman