hamishcoleman/thinkpad-ec #40: Revert patched Bios

vldn-de, 2017-08-19:
Hey, is it possible to revert the bios patch that i can use the original x230 Keyboard in my x230 again?

Best regard and thanks for the tool!

diggit, 2017-08-19:
Hi,
from README:
> * To include the battery validation patch or to make a build that
  reverts any EC changes, read the docs/CONFIG.txt and follow
  the configuration instructions in it before running step 6.

Here you can find [CONFIG.txt](https://github.com/hamishcoleman/thinkpad-ec/blob/master/docs/CONFIG.txt)

> Any combination of the two sets of patches can be enabled or disabled
(even including a version with no patches at all - to revert all changes)

vldn-de, 2017-08-19:
even if i had installed the newest bios over the patched one?

Thx for the fast answer!!

diggit, 2017-08-19:
If flashing newest BIOS flashed EC too, all patches are reverted.

jeremyweston, 2017-09-24:
This worked OK for me with my T530 I'm prepping to resell.

I reverted the keyboard and right before step 6, I simply ran
`make patch_disable_keyboard clean`

then continued on with step 6.  I know the output is a bit gabbled, but it does alter the scripts to disable the keyboard EC flash in the img/iso file output.

I tested this with a T430 too but it didn't work. However, this may be due to a faulty keyboard someone sold me. I'm going to swap another one in and see if it worked with the T430 or if there was a bug.

jeremyweston, 2017-09-24:
The backlighting isn't working on the T430 BIOS reflash. I tried it with a known working keyboard with a known working backlight. Didn't have the backlight on the T430.

hamishcoleman, 2017-09-25:
@jeremyweston did you do the steps to isolate the pins on your x220 keyboard before plugging it into your x230?  It is possible that the over-current condition that results from not isolating it will blow a fuse on the motherboard - stopping the backlight from working.

I cannot check or test this as I have no backlit keyboards.

jeremyweston, 2017-09-25:
@hamishcoleman,

This error was produced with a T430, not a X230. 

I actually do own a X230 as well and was able to successfully add the x220 keyboard to it thanks to your work. (Thank you!)

However, the issue I've come across involves the T430 keyboard on a T430 not having a backlight after using the disable keyboard make files before generating a BIOS to flash.

` make patch_disable_keyboard clean`

I did this with a T430 and a T530. It worked great for the T530. But the BIOS for the T430 is not restoring the backlight function on the T430. 

When I press FN+Spacebar, nothing happens. If I do it again, nothing happens again, if I do it a third time, the ThinkLight comes on. Then this cycle repeats.

jeremyweston, 2017-09-25:
I swapped the keyboards between the T530 and the T430 to make sure it wasn't a hardware issue. Both keyboards work OK.

hamishcoleman, 2017-09-25:
Sorry, I misspoke - when I said "x230" above, I meant "t430" and when I said "x220", I meant "t420"

Did you place a t420 keyboard in your t430?
Did you isolate the pins? (as shown in http://www.thinkwiki.org/wiki/Install_Classic_Keyboard_on_xx30_Series_ThinkPads#Pinout )

explit7, 2017-12-13:
i can also confirm: Reverting the patch back on the ThinkPad T430 doesn't work. Thinklight is still not on Fn + Space.
That would be very good if the patch would be reversible on the T430.

explit7, 2018-01-20:
problem seems to be solved. Patching or revert patch back requires the battery and power adapter connected!
without power adapter connected, the EC will not be written. you will also not get an wsrning message or an error. it just boot normal.

hamishcoleman, 2018-02-05:
Good to hear @explit7, I have added some more words to the README to suggest plugging the power adapter in.

closed 2018-02-05 by hamishcoleman