hamishcoleman/thinkpad-ec #41: Battery Patch not working

Jabbion, 2017-08-20:
I have run
`make patch_enable_battery clean`
`make patched.t430.img`
`sudo dd if=patched.t430.img of=/dev/sdc`

from make patched.t430.img the output was:
```
Patches applied:
t430.G1HT35WW.img.d/001_keysym.patch t430.G1HT35WW.img.d/002_dead_keys.patch
t430.G1HT35WW.img.d/003_keysym_replacements.patch
t430.G1HT35WW.img.d/004_fn_keys.patch t430.G1HT35WW.img.d/005_fn_key_swap.patch
t430.G1HT35WW.img.d/006_battery_validate.patch
```
but still when booting with the usb-Stick, I get something like this:
Patches applied:
```
t430.G1HT35WW.img.d/001_keysym.patch t430.G1HT35WW.img.d/002_dead_keys.patch
t430.G1HT35WW.img.d/003_keysym_replacements.patch
t430.G1HT35WW.img.d/004_fn_keys.patch t430.G1HT35WW.img.d/005_fn_key_swap.patch
```
Any ideas how to apply this patch when booting with the USB-Stick?

hamishcoleman, 2017-08-21:
First, we can confirm your build status - can you paste the patched.t430.iso.report file?

Second, if you look at the files on the usb stick, what is the sha1sum of the /FLASH/G1HT35WW/$01D2000.FL2 ?

After building with the same settings as you, I get a FL2 file with the checksum 890c86193fd415bb1476fff859c567b1e62842a3

Jabbion, 2017-08-21:
I do not have the Directory  /FLASH/**G1HT35WW**. A Directory similar as G1HT35WW is `G1ETB2WW`. The sha1sum of **G1ETB2WW/$01D2000.FL2** is

`2f49d711d520575ff9079d74c64ef495850a9818`
[did not expected to be the same as you posted because different file]

In the patched.t430.iso.report file is:
`Buildinfo: v1-174-g9d5135-dirty (20170820) patched.t430.img
Built: 890c86193fd415bb1476fff859c567b1e62842a3  t430.G1HT35WW.s01D2000.FL2

Patches applied:
```
t430.G1HT35WW.img.d/001_keysym.patch t430.G1HT35WW.img.d/002_dead_keys.patch
t430.G1HT35WW.img.d/003_keysym_replacements.patch
t430.G1HT35WW.img.d/004_fn_keys.patch t430.G1HT35WW.img.d/005_fn_key_swap.patch
t430.G1HT35WW.img.d/006_battery_validate.patch

Description: t430 BIOS 2.72 (G1ETB2WW) EC 1.13 (G1HT35WW)
```

hamishcoleman, 2017-08-23:
Sorry, I was rushing and used the wrong filename - the file you have checked is the same one I was checking.

The checksum you have is the one you would get without the battery patch - and that also means that you are in a state that the build system is not expecting (since patched.t430.iso.report file is reporting that the battery patch was, in fact, applied - there is clearly something unexpected happening)

Can you also check the sha1sum on "t430.G1HT35WW.s01D2000.FL2" ?

It might be worth trying a "make really_clean" followed by a "make patched.t430.img" and checking the build output for error messages

Jabbion, 2017-08-24:
After running `make really_clean` and make `patched.t430.img` and it worked!

Thank you so much!

hamishcoleman, 2017-08-27:
Good news!

closed 2017-08-27 by hamishcoleman