hamishcoleman/thinkpad-ec #46: .FL1 file support

skochinsky, 2017-08-31:
You mention in readme problems with the L530  update. I downloaded it and it sems it has the $PFH trailer which contains names and offsets of the regions included in the file  (in the update I downloaded it's at offset `008001D0`). I documented the layout [here](https://gist.github.com/skochinsky/181e6e338d90bb7f2693098dc43c6d54) and you can find a sample implementation in https://github.com/theopolis/uefi-firmware-parser/commit/aea0eaf4608958fed86e68e92d9aeaf17dbfd16f

skochinsky, 2017-08-31:
Correction: it seems this `FL1` is a capsule and the flash image starts at 0x1D0 bytes. After stripping the header, the data from `$PFH` header lines up properly:

```
4 images at 008000DC
00000000 00001000 Desc
00001000 003FF000 Me
00400000 00020000 Ec
00420000 003E0000 Bios
```

hamishcoleman, 2017-09-04:
Cool, thanks - I'll add some slicing for the $PFH header and see if there is anything useful inside there

hamishcoleman, 2017-09-05:
I've added support to extract the EC image from a FL1 file with a $PFH header, but unfortunately the firmware is for a different CPU architecture, so the existing patch recipes cannot be simply ported.

closed 2018-02-05 by hamishcoleman