hamishcoleman/thinkpad-ec #47: Mod + latest x230 bios causes EC freeze?

flammit, 2017-09-12:
I checked out a clean repo to build the EC update and it's using G2ETA8WW / g2uj25us.exe. Has anyone else experienced this problem where the x230 will eventually not respond to the power button?

The lockup happens only after the battery is plugged in for a short amount of time.  Holding the power button for 20-30 seconds w/o battery and AC to do the EC reset fixes things temporarily until the next hang, so it is a bit annoying.

I've reverted to using an older bios version (G2ETA6WW / g2uj23us.exe) to build the EC update image and this seems works fine.

hamishcoleman, 2017-09-12:
That sounds bad!

Can you confirm what BIOS version you have installed?

There is no EC code changes between G2ETA6WW and G2ETA8WW - so any issue must be coming from the BIOS version itself.

flammit, 2017-09-12:
The BIOS was on G2ETA8WW with ME untouched. As expected, the EC freeze happened regardless of the BIOS installed at the time - I experienced freeze with both stock BIOS and with coreboot.

As I said, things are working fine now, but was just curious if others experienced this or could repro. Not sure how to debug this, but happy to try with some guidance.

hamishcoleman, 2017-09-19:
I have not heard of any other similar issues, so that doesnt add any info.

I did try building a patched firmware with both the old and the new bios as source and got the exact same firmware file built both times - so, unless there was an issue with your build process, that leaves the BIOS - which we dont have a lot that we can debug

flammit, 2017-09-19:
Spoke too soon - EC locked up again.  Looks like I might have a hardware problem unrelated to this update as I also recently experienced some kind of short or sizzling overcharge on the battery connector :/  As a last ditch effort I'm going to try to replace the CMOS, but I'm not hopeful.

Sorry for the FUD.  I'm guessing you can probably revert your change.

closed 2017-09-23 by flammit