hamishcoleman/thinkpad-ec #52: X230 bios 2.68

Polishapple, 2017-09-28:
Hi, there is new bios update on lenovo site

vatrat, 2017-10-04:
I just got this laptop. Would installing this bios update break this patch?

hamishcoleman, 2017-10-09:
The X230 v2.68 BIOS contains the same EC version as the previous couple of BIOS's - so installing the bios update should not overwrite the EC patch (and if it did, you can simply re-install the patch)

That being said, the support for building against the 2.68 bios has been available since June and was only disabled recently due to a report of an issue - we no think that report is unrelated to this patch.

vatrat, 2017-10-09:
I went to the site where you can get BIOS upgrades and used 2.66 anyway, just in case (since building the patch said it was for version 2.66), and it worked.

hamishcoleman, 2018-04-14:
Current build has been updated to x230 BIOS 2.70

closed 2018-04-14 by hamishcoleman