hamishcoleman/thinkpad-ec #53: How to undo a T430 mod?

sebileis2017, 2017-10-09:
Hi there, I was wondering how I can undo the mod on a T430? I'm selling my T430 and was planning on putting the original keyboard back in, is it possible to undo the mod so the correct function keys for the T430 layout are recognized?

hamishcoleman, 2017-10-09:
There sure is - you can build the firmware with all the patches disabled.

Have a read of https://github.com/hamishcoleman/thinkpad-ec/blob/master/docs/CONFIG.txt and see if that answers your question

sebileis2017, 2017-10-09:
Do I build this firmware the same way I built the mod firmware in Linux?

hamishcoleman, 2017-10-09:
Yes, it is the same firmware and build system, with some options changed to remove the patches

sebileis2017, 2017-10-09:
Okay, thanks. I will give this a try tomorrow.

sebileis2017, 2017-10-09:
I will comment here if it works or not.

explit7, 2017-12-15:
I tryed on 2x Thinkpad T430. Keyboard mod is ok, but reverting doesn't work.
Did someone find the way to unapply patch?

explit7, 2017-12-23:
This is very strange. I build new package without all patches at all (it says original Lenovo EC Package)
Tha package is flashed on the T430 but after restart the Thinklight is still on Fn+PgUp (Classic keyboard) and not FN+Space (Like it should be on new keyboard).

Alredy tried on 2x ThinkPad T430. Patch works. Reverting back doesn't work. Latest BIOS 2.74 + EC 1.13

explit7, 2018-01-20:
can be closed. patching or undo patch requires battery and AC-Adapter connected!

closed 2018-02-05 by hamishcoleman