hamishcoleman/thinkpad-ec #55: x230 BIOS 2.70

ccarlile, 2017-10-20:
I have my thinkpad updated to the latest version. The bootable image that was made automatically wouldnt proceed - BIOS flash utility said it didn't match the hardware. The bios ISO is `g2uj26us.iso`. I can take a whack at it if I have an outline of the procedure.

hamishcoleman, 2017-10-20:
The process is pretty simple.  Open the Descriptions.txt and find the four occurrences of "g2uj23us.iso" and replace all but the first one with "g2uj26us.iso"

However, nothing in the update image flashing process has changed (except for the BIOS firmware file itself), so it seems strange that there would be an issue like you are describing.

I know it sounds like a silly question, but are you sure you have a x230?

ccarlile, 2017-10-20:
It does say x230 on the monitor, and it is not the tablet version. I'll give it a whack this evening.

rslindee, 2017-10-21:
Per the 2.70 release notes:

`(Note) If the UEFI BIOS has been updated to version 2.70 or higher, it is no longer able to roll back to the version before 2.70 for security improvement.`

I think this may be causing your problem. I'm experiencing a similar issue trying to downgrade my X230 BIOS from 2.70 an earlier version.

hamishcoleman, 2017-10-21:
@rslindee - if that was the case, I would expect a message like "Signed capsule only allow equal and newer version." and not "BIOS part numbers do not match".  Also, there is the point that this patch is not flashing the BIOS, so does not run up against the BIOS version downgrade requirements.

WarhawkCZ, 2017-11-19:
First of all, thanks a lot for all the effort you guys spend on this project.
I am having issues with the EC controller flashing. I followed the manual too much and updated my BIOS to version 2.70. However, I am not able to get a bootable ISO from the scripts. The verbose output seems to be OK but the bios flashing utility never shows up no matter how I try. Just a black screen pop-ups and then it goes back to the boot selection option. 
When I take just the original ISO from Lenovo, I am able co create a bootable disk and enter the utility without any problem. That's how I upgraded my bios to 2.70 before. I ran out of ideas what else to try. BIOS downgrade is not possible.

Does anyone has an idea what else to try?

WarhawkCZ, 2017-11-20:
OK, so I have an update. Today I made the USB drive bootable. I can't that completely comprehend what was wrong but t seems that `sudo dd if=patched.x230.img of=/dev/sdx` does not make the job for me (Linux Mint 18.2). That's strange because when I replace the pathched.x230.img file by original BIOS image (after ISO->IMG) it works and boots.
It may be possible that both USB sticks I used contained other fragments in memory which corrupted the process. I simply don't get it.
I used YUMI - Multiboot tool, selected GRUB from the menu & FAT formatting. That worked immediately.

Tip: If you're not sure what is wrong, create a virtual machine (Virtual Box) and try to boot the ISO. If it works in the virtual machine, you're probably messing up crating a bootable USB stick.

naivePC, 2018-02-04:
I am naive user, I have a x230 laptop and my battery drained out completely. Now I bought third party battery and as suggested in this community and by seeing the youtube video I tried to get rid of battery white list. I successfully followed the steps till patch file turned ON. after that when I tried to make patch I got error message after typing "make patch.x230.img" ultimately saying failed to make patch image file.
Please help me...

naivePC, 2018-02-04:
actual message as follows
ownloading x230 BIOS 2.66 (G2ETA6WW) EC 1.14 (G2HT35WW)
wget -O g2uj23us.iso.orig https://download.lenovo.com/pccbbs/mobiles/g2uj23us.iso
--2018-02-04 08:07:16--  https://download.lenovo.com/pccbbs/mobiles/g2uj23us.iso
Resolving download.lenovo.com (download.lenovo.com)... 23.35.220.108
Connecting to download.lenovo.com (download.lenovo.com)|23.35.220.108|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 33927168 (32M) [application/octet-stream]
Saving to: ‘g2uj23us.iso.orig’

g2uj23us.iso.orig   100%[===================>]  32.36M   798KB/s    in 38s     

2018-02-04 08:07:54 (861 KB/s) - ‘g2uj23us.iso.orig’ saved [33927168/33927168]

scripts/checksum --rm_on_fail g2uj23us.iso.orig g2uj23us.iso
touch g2uj23us.iso.orig
./scripts/ISO_copyFL2 from_iso g2uj23us.iso.orig x230.G2HT35WW.s01D3000.FL2.orig 01D3000.FL2
git submodule update --init --remote

I think iso mentioned in the message belong to older bios but today only I have updated to 2.7 version BIOS.

is their any solution for this error?

naivePC, 2018-02-04:
Even I tried after changing the file name as suggested "g2uj26.iso' but still not working.

rad4day, 2018-02-05:
Well I don't see any error message in your submission to be honest 🤔

naivePC, 2018-02-05:
Oh sorry I forgot to paste last 4 lines. Yes its true I did get error message.  After typing "make patched.x230.img" it processed but ultimately got failed to create iso image. As I have little knowledge about the programming I could not explain in technical manner sorry for that. Anyway, I tried as described and followed each step carefully but without success. My BIOS version is 2.70.
Problem raised regarding BIOS version in earlier post by CCarilile and hamishcoleman suggested Open the Descriptions.txt and find the four occurrences of "g2uj23us.iso" and replace all but the first one with "g2uj26us.iso". 
I did exactly but without success. Even I replaced latest checksum 2.70 version BIOS. I think I might have doing minor error which I could not identify.
Can anyone explain what is the exact meaning of " replace all but the first one with "g2uj26us.iso" mentioned by hamishcoleman. Is it just replacing the name of the file or replace all the description regarding 2.70 version? 
I do not know much about programming so if someone can paste the entire description.txt file for x230 2.70 BIOS  will be helpful. I will try again. I will till till I get success.

hamishcoleman, 2018-02-05:
To be very clear to anyone reading this, my suggestions to update the Descriptions.txt file were /only/ if you were intending to create an update to this repo to add support for new firmware versions.

If you are simply trying to install the patched firmware, then the 2.70 bios uses the same EC as the previous three releases and any issues are unrelated to the BIOS versions

hamishcoleman, 2018-02-05:
@naivePC you sound like you have a completely different issue to what this ticket is for - can you open a new ticket and paste every single line that your build produces into it.  There are important debugging details that are included in the build output that are needed to do any diagnostics

naivePC, 2018-02-05:
here is the full description of my trial
First problem - I could not download the files as shown in the video

mint@mint ~ $ sudo apt-get install build-essential git mtools libssl-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
mtools is already the newest version (4.0.18-2ubuntu0.16.04).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 build-essential : Depends: g++ (>= 4:5.2) but it is not going to be installed
 libssl-dev : Depends: libssl1.0.0 (= 1.0.2g-1ubuntu4) but 1.0.2g-1ubuntu4.8 is to be installed
              Depends: zlib1g-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

So to overcome the problem I have downloaded from git and continued like this

mint@mint ~ $ ls
Desktop    Downloads  Pictures  Templates           Videos
Documents  Music      Public    thinkpad-ec-master
mint@mint ~ $ cd thinkpad-ec-master
mint@mint ~/thinkpad-ec-master $ make list_laptops
./scripts/generate_deps Descriptions.txt >.d/generated.deps

The following make targets are the supported usb images:

patched.t430.img    - for patching Thinkpad T430
patched.t430s.img    - for patching Thinkpad T430s
patched.t530.img    - for patching Thinkpad T530
patched.t530i.img    - for patching Thinkpad T530i
patched.w530.img    - for patching Thinkpad W530
patched.x230.img    - for patching Thinkpad X230
patched.x230t.img    - for patching Thinkpad X230t
mint@mint ~/thinkpad-ec-master $  make patch_disable_keyboard clean
for j in 001_keysym.patch 002_dead_keys.patch 003_keysym_replacements.patch 004_fn_keys.patch 005_fn_key_swap.patch; do \
         for i in *.img.d; do if [ -e $i/$j ]; then mv $i/$j $i/$j.OFF; fi; done; \
done
rm -f .d/generated.deps \
            patched.*.iso patched.*.img *.FL2 *.FL2.orig *.img.enc \
            *.img.enc.orig *.img.orig *.bat *.report \
            *.img \
            *.txt.orig
rm -rf *.iso.extract *.iso.orig.extract
mint@mint ~/thinkpad-ec-master $ make patch_enable_battery clean 
./scripts/generate_deps Descriptions.txt >.d/generated.deps
for i in *.img.d; do if [ -e $i/006_battery_validate.patch.OFF ]; then mv $i/006_battery_validate.patch.OFF $i/006_battery_validate.patch; fi; done
rm -f .d/generated.deps \
            patched.*.iso patched.*.img *.FL2 *.FL2.orig *.img.enc \
            *.img.enc.orig *.img.orig *.bat *.report \
            *.img \
            *.txt.orig
rm -rf *.iso.extract *.iso.orig.extract
mint@mint ~/thinkpad-ec-master $ make patch_enable_battery clean 
./scripts/generate_deps Descriptions.txt >.d/generated.deps
for i in *.img.d; do if [ -e $i/006_battery_validate.patch.OFF ]; then mv $i/006_battery_validate.patch.OFF $i/006_battery_validate.patch; fi; done
rm -f .d/generated.deps \
            patched.*.iso patched.*.img *.FL2 *.FL2.orig *.img.enc \
            *.img.enc.orig *.img.orig *.bat *.report \
            *.img \
            *.txt.orig
rm -rf *.iso.extract *.iso.orig.extract
mint@mint ~/thinkpad-ec-master $ make patched.x230.img
./scripts/generate_deps Descriptions.txt >.d/generated.deps
Downloading x230 BIOS 2.66 (G2ETA6WW) EC 1.14 (G2HT35WW)
wget -O g2uj23us.iso.orig https://download.lenovo.com/pccbbs/mobiles/g2uj23us.iso
--2018-02-05 10:48:24--  https://download.lenovo.com/pccbbs/mobiles/g2uj23us.iso
Resolving download.lenovo.com (download.lenovo.com)... 23.35.220.108
Connecting to download.lenovo.com (download.lenovo.com)|23.35.220.108|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 33927168 (32M) [application/octet-stream]
Saving to: ‘g2uj23us.iso.orig’

2018-02-05 10:48:45 (1.59 MB/s) - ‘g2uj23us.iso.orig’ saved [33927168/33927168]

scripts/checksum --rm_on_fail g2uj23us.iso.orig g2uj23us.iso
touch g2uj23us.iso.orig
./scripts/ISO_copyFL2 from_iso g2uj23us.iso.orig x230.G2HT35WW.s01D3000.FL2.orig 01D3000.FL2
git submodule update --init --remote
/bin/bash: git: command not found
Makefile:272: recipe for target 'mec-tools/Makefile' failed
make: *** [mec-tools/Makefile] Error 127
mint@mint ~/thinkpad-ec-master $ 

Ultimately showing error 127. This is exactly whats happening to me. Please help me.

hamishcoleman, 2018-02-05:
@naivePC this would be much simpler if you opened a new ticket for your different problem.

You do not have git installed - this software is essential, so you will need to install that first.

majekw, 2018-04-08:
I can confirm that replacing g2uj23us.iso to g2uj26us.iso in Descriptions.txt works on X230 with bios 2.70. At least for me :-)

hamishcoleman, 2018-04-14:
Thanks @majekw, I've applied that patch

closed 2018-04-14 by hamishcoleman

hamishcoleman, 2018-10-08:
>    Does this work with BIOS 2.73? I have installed it with g2uj29us.iso

Yes, it does work - the current version is building against g2uj29us.iso
and Lenovo has not made any changes to the EC firmware for a while now.

finiteautomata, 2018-10-08:
Thank for you answer @hamishcoleman; I deleted my comment as it wasn't related to this and opened a new issue (#94)