hamishcoleman/thinkpad-ec #56: X230 EC Won't Flash with 2.70 BIOS

rslindee, 2017-10-21:
I followed the guide, created a bootable USB stick and booted from it. Once booted, I got the text saying this was my patched X230 BIOS and received the "Press any key to flash" alongside a bunch of text indicating this was a patched image.

I received a notice that the update occurred successfully (this was extremely quick and much shorter than a typical BIOS update). After that I can restart the machine, but I don't get any secondary "flashing update" text that is mentioned in the guide I've followed. Additionally, the keyboard layout doesn't appear to have changed when I boot into the OS. I've tried rebooting several times, cold power off/on, and even flipping the "UEFI/Legacy" boot mode setting.

This is all using the 2.70 BIOS. Has anyone had any success patching the EC firmware from this BIOS version? One thing that has me worried is the notice on the Thinkpad support site:

`(Note) If the UEFI BIOS has been updated to version 2.70 or higher, it is no longer able to roll back to the version before 2.70 for security improvement.`

Could this (rather recent) BIOS update be affecting the ability to patch the EC firmware?

hamishcoleman, 2017-10-21:
The first thing that I can tell you is that the BIOS version and updates are not the same as the EC firmware version and updates - this is why the EC flashing is much faster than you have seen for your BIOS flashing.  Also, the EC firmware is not a downgrade - they have not changed the EC version for the last 10 update packages - so the UEFI BIOS downgrade warnings are definitely not applicable.

It is possible that they have changed the way the BIOS handles a request for an EC upgrade - but the official flash tools have not been changed at all between these versions, so it is unlikely.

Finally - while unfortunately it doesn't help you - you are probably suffering from the same thing that people have been people with completing the flashing process since the project started.  We don't really know how the Lenovo code works internally, so we end up just trying lots of combinations (like you have done with reboots and BIOS option settings) trying to figure out what change makes it suddenly work.

rslindee, 2017-10-21:
@hamishcoleman Thank you for the quick reply, your insight is greatly appreciated.

After trying a few more combinations, I tried using the Windows BIOS update utility to clear the model number. It seemed that immediately after the reboot I got the "updating" message and after that the keyboard layout appears to be updated! I'm really puzzled why this made a difference. Perhaps clearing the model number triggered an update for the firmware controller?

closed 2017-10-21 by rslindee

hamishcoleman, 2017-10-21:
Good news!  I've never used the Windows BIOS update utility (since I dont use Windows) - what does is say that the "clear the model number" is supposed to do??  (I'd be surprised if it actually cleared the model number, I cannot see the software suddenly all reporting that you had a "Thinkpad $NOTHING" !!)

rslindee, 2017-10-21:
@hamishcoleman I'm really not sure what the purpose of said "model number" feature is, either (it only allows you to change/clear the last three characters of said number, I might add).

Likewise, I'm not a Windows user either (this machine happened to come preloaded), but this may prove to be a useful troubleshooting/data point for future reference.

hamishcoleman, 2017-10-21:
Yes - I am thinking of future troubleshooting.  I'll be trying to add something about this to the list of things to try.

Thanks!