hamishcoleman/thinkpad-ec #57: t430s: BIOS 2.70/2.67 seem to result in unresponsive keyboard

four0four, 2017-11-12:
I've tried both 2.70 and 2.67, it seems that the keyboard is utterly unresponsive (aside from the power switch and power/mute LEDs) once the patch is applied. Disappointingly, applying an unpatched version does not seem to fix this issue. 

My board happily claims to apply to EC update, including the reboot and "applying EC firmware" stage...

This really feels like a hardware problem, but the only thing I did before the failure was apply this patch. If it helps, I can manually reflash to 2.6x (2.70 prevents software-based downgrades) and attempt again, but I figured I'd open an issue in order to see if anyone else has had a similar problem first.

four0four, 2017-11-20:
Quick update: I got the MEC talking over JTAG, so I'l be able to validate the contents of the flash relatively soon / determine if something else broke. 

If there's interest, I can also document how to do this.

cmorche, 2017-11-23:
I am very interested in this! I've bricked several T430S motherboards while messing with this, Coreboot and the full HD panel mod. I'd love to know how to use JTAG to possibly recover some of those systems.

Thank you!

four0four, 2017-11-23:
Awesome - I'll try and get some stuff packaged up nicely in the next few days. I don't think it's legal to distribute the images directly, but I can detail the procedure well enough (spoiler: it's identical to the flashed image, minus the lower word of the checksum at the end...not sure why). At the minimum, you'll need some a JTAG adapter with a working OpenOCD config, a steady hand, and a soldering iron :)

Feel free to shoot me an email direct if you want to discuss particulars / confirm anything.

hamishcoleman, 2017-11-23:
Just adding my encouragement to that of cmorche - yes, this sounds like a great thing to have documented.  It finally gives us an answer to "I've bricked my laptop, now what".

I'd like to add details on the JTAG attachement points and the OpenOCD to the docs already in the repo - and at very least document how to make the image (probably not adding a makefile target at this stage, but that is a possibility too)

So - if you dont mind - once you have written it up, I can add a copy of your docs to the repo.

four0four, 2017-12-08:
Good to hear :)

I actually fiddled around a bit, and it started working...which is very unsatisfying. I've got half a writeup done, but it turns out that Synopsys doesn't specify a standard flash interface and as such their OpenOCD port doesn't have any sense as to how to talk to the flash IP Microchip/SMSC are using. I'll release docs once I've got code which fixes this (this may be after or during CCC, depending).

In the mean time, [here's](https://gist.github.com/four0four/680e1fa70e7c216baac2afbd459d03d8) the OpenOCD config, and [here's](https://i.imgur.com/Z0DlbjE.jpg) the JTAG pinout. Also of mild interest might be that I've dumped the bootrom as well...and it refers to itself as the MEC16_18_, so that further corroborates the theory that they're basically the same chip.

closed 2017-12-13 by four0four