hamishcoleman/thinkpad-ec #60: T430s new BIOS 2.69

T-Lancer, 2017-12-28:
Lenovo has recently released a new BIOS for the T430s.  2.69.  this replaces the 2.67 version I currently have running on my Thinkpad.  It seems it includes new Embedded Controller Firmware.  I have the battery validator check patched and this update would probably remove that.  Can we see an updated script for 2.69 for the T430s?

thank you.

https://pcsupport.lenovo.com/de/en/products/LAPTOPS-AND-NETBOOKS/THINKPAD-T-SERIES-LAPTOPS/THINKPAD-T430S/downloads/DS029725

explit7, 2017-12-28:
Hello.
As i know 2.67 and 2.69 include the same version of embedded controller.

T-Lancer, 2017-12-28:
indeed seems that way.  a colleuge mistakingly installed 2.69 without checking, but battery check is still disabled.  should be OK to flash if you already have 2.67 flashed with modded EC firmware.

closed 2017-12-28 by T-Lancer