hamishcoleman/thinkpad-ec #62: Flash goes through on X230 but keys are the same

munderwoods, 2018-01-15:
I constructed the image following the instructions to the letter in arch linux. I then enabled legacy boot mode in order to boot the drive. It booted and flashed everything fine. It showed the patch in there before it flashed. I've done this about three times now. I switched back to UEFI mode afterword and rebooted several times. My keys did not change. Is there something I am missing or something I can check? I'm on bios version 2.7.

munderwoods, 2018-01-17:
It was my damn charger. The charger was the legit one that came with the laptop and was charging but it would not flash while charging. I used my generic one and that did the trick!

closed 2018-01-17 by munderwoods

hamishcoleman, 2018-01-17:
Hey there - thanks for being persistent and getting it working.  Even more thanks for coming back and updating with all the details.

Sorry that I was too busy to get back to you sooner.

munderwoods, 2018-01-17:
Thanks for making such a great app!