hamishcoleman/thinkpad-ec #65: Instruction wrong?

phandojo, 2018-01-27:
Says here that a T420 palmrest can be used: https://www.thinkwiki.org/wiki/Install_Classic_Keyboard_on_xx30_Series_ThinkPads#The_replacement_.28T430.29

But a video contradicting the info can be found here: https://www.youtube.com/watch?v=V74pSO8of4M

I am about to buy a 7-row keyboard for T430, but unsure if I should get anything else.

Thoughts?

Thanks in advance.

explit7, 2018-01-28:
Palmrest from the T420 can't be used for T430. Only X220 palmrest instead X230 Palmrest.

You need to modify the t420 keyboard, so it can fit in T430.

hamishcoleman, 2018-01-28:
Thanks @explit7 - I've never had a t430 to try the mod on myself, so it is good to have confirmation.  I've updated that thinkwiki section with the details.

phandojo, 2018-01-28:
@hamishcoleman I have T430 coming in and will use the 7-row keyboard

This guide have info on fixing the top portion of KB to prevent T430 chassis bulge, but not sure how accurate it is: http://www.instructables.com/id/ThinkPad-T430T430sX230-Classic-Keyboard-Mod/

hamishcoleman, 2018-02-01:
@phandojo Unfortunately, I dont know how accurate that instructable is either - I've never actually had a T430 to try to mod.

The cutting off of the plastic nubs to stop the bulge sounds like it would work, but you should experiment

closed 2019-08-06 by hamishcoleman