hamishcoleman/thinkpad-ec #66: X230 whitelist thirdparty battery

naivePC, 2018-02-06:
As described the problem previously in ticket x230 BIOS 2.70
I continued my try to patch my bios 

As suggested by Hamishcoleman I have installed git in my portable USB still I could not get the completed iso patch file. Here is the message shown 

mint@mint ~/thinkpad-ec $ make patched.x230.img
./scripts/generate_deps Descriptions.txt >.d/generated.deps
Downloading x230 BIOS 2.66 (G2ETA6WW) EC 1.14 (G2HT35WW)
wget -O g2uj23us.iso.orig https://download.lenovo.com/pccbbs/mobiles/g2uj23us.iso
--2018-02-06 01:59:32--  https://download.lenovo.com/pccbbs/mobiles/g2uj23us.iso
Resolving download.lenovo.com (download.lenovo.com)... 23.35.220.108
Connecting to download.lenovo.com (download.lenovo.com)|23.35.220.108|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 33927168 (32M) [application/octet-stream]
Saving to: ‘g2uj23us.iso.orig’

g2uj23us.iso.orig   100%[===================>]  32.36M  2.01MB/s    in 19s     

2018-02-06 01:59:51 (1.69 MB/s) - ‘g2uj23us.iso.orig’ saved [33927168/33927168]

scripts/checksum --rm_on_fail g2uj23us.iso.orig g2uj23us.iso
touch g2uj23us.iso.orig
./scripts/ISO_copyFL2 from_iso g2uj23us.iso.orig x230.G2HT35WW.s01D3000.FL2.orig 01D3000.FL2
git submodule update --init --remote
Submodule 'mec-tools' (https://github.com/eigenmatt/mec-tools) registered for path 'mec-tools'
Cloning into 'mec-tools'...
remote: Counting objects: 29, done.
remote: Total 29 (delta 0), reused 0 (delta 0), pack-reused 29
Unpacking objects: 100% (29/29), done.
Checking connectivity... done.
Submodule path 'mec-tools': checked out '07a1b1407326488376afed6c596d37c53d7432b6'
git submodule update
make -C mec-tools
make[1]: Entering directory '/home/mint/thinkpad-ec/mec-tools'
cc -Wall -O2 mec_csum_boot.c -o mec_csum_boot 
mec_csum_boot.c:5:19: fatal error: stdio.h: No such file or directory
compilation terminated.
Makefile:8: recipe for target 'mec_csum_boot' failed
make[1]: *** [mec_csum_boot] Error 1
make[1]: Leaving directory '/home/mint/thinkpad-ec/mec-tools'
Makefile:275: recipe for target 'mec-tools/mec_encrypt' failed
make: *** [mec-tools/mec_encrypt] Error 2

hamishcoleman, 2018-02-06:
Looks like your build has made some progress.  However, you still appear to be missing some required software.

Are you following the Step-by-step instructions from the README file?  In step 1, it has a command to install all the needed packages :-
```
apt-get install build-essential git mtools libssl-dev
```

Once all those packages are installed, you should be able to perform a complete build

naivePC, 2018-02-06:
Yes, I tried it is showing 

Do you want to continue? [Y/n] Y
Ign:1 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 git-man all 1:2.7.4-0ubuntu1.1
Get:2 http://archive.ubuntu.com/ubuntu xenial/main amd64 liberror-perl all 0.17-1.2 [19.6 kB]
Ign:3 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 git amd64 1:2.7.4-0ubuntu1.1
Err:1 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 git-man all 1:2.7.4-0ubuntu1.1
  404  Not Found [IP: 91.189.91.26 80]
Err:3 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 git amd64 1:2.7.4-0ubuntu1.1
  404  Not Found [IP: 91.189.91.26 80]
Fetched 19.6 kB in 1s (14.4 kB/s)
E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/g/git/git-man_2.7.4-0ubuntu1.1_all.deb  404  Not Found [IP: 91.189.91.26 80]

E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/g/git/git_2.7.4-0ubuntu1.1_amd64.deb  404  Not Found [IP: 91.189.91.26 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

naivePC, 2018-02-06:
I have tried in both ubuntu and linux mint but no success. Do I need to install full version in my laptop?

hamishcoleman, 2018-02-06:
Sounds like the apt index needs updating

You should try running the suggested ```apt-get update``` before trying to install the packages

naivePC, 2018-02-06:
Git package seems installed but upadate was not successful. full description as follows,

mint@mint ~ $ sudo apt-get install git-all
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  apache2 apache2-bin apache2-data apache2-utils cvs cvsps emacs emacs24
  emacs24-bin-common emacs24-common emacs24-common-non-dfsg git git-arch
  git-cvs git-doc git-el git-email git-gui git-man git-mediawiki git-svn gitk
  gitweb libapr1 libaprutil1 libaprutil1-dbd-sqlite3 libaprutil1-ldap
  libclass-factory-util-perl libclass-singleton-perl
  libdatetime-format-builder-perl libdatetime-format-iso8601-perl
  libdatetime-format-strptime-perl libdatetime-locale-perl libdatetime-perl
  libdatetime-timezone-perl libdbd-sqlite3-perl libdbi-perl liberror-perl
  libjson-perl liblua5.1-0 libm17n-0 libmediawiki-api-perl
  libmodule-implementation-perl libmodule-runtime-perl libotf0
  libpackage-deprecationmanager-perl libpackage-stash-perl
  libparams-classify-perl libparams-util-perl libparams-validate-perl
  libserf-1-1 libsub-install-perl libsvn-perl libsvn1 libterm-readkey-perl
  libtry-tiny-perl libyaml-perl m17n-db tla
Suggested packages:
  apache2-doc apache2-suexec-pristine | apache2-suexec-custom mksh rcs
  ncurses-term git-daemon-run | git-daemon-sysvinit mediawiki subversion
  libmldbm-perl libnet-daemon-perl libsql-statement-perl m17n-docs
  libscalar-number-perl libyaml-shell-perl
Recommended packages:
  emacs24-el git-daemon-run | git-daemon-sysvinit libjson-xs-perl
  libpackage-stash-xs-perl tla-doc
The following NEW packages will be installed:
  apache2 apache2-bin apache2-data apache2-utils cvs cvsps emacs emacs24
  emacs24-bin-common emacs24-common emacs24-common-non-dfsg git git-all
  git-arch git-cvs git-doc git-el git-email git-gui git-man git-mediawiki
  git-svn gitk gitweb libapr1 libaprutil1 libaprutil1-dbd-sqlite3
  libaprutil1-ldap libclass-factory-util-perl libclass-singleton-perl
  libdatetime-format-builder-perl libdatetime-format-iso8601-perl
  libdatetime-format-strptime-perl libdatetime-locale-perl libdatetime-perl
  libdatetime-timezone-perl libdbd-sqlite3-perl libdbi-perl liberror-perl
  libjson-perl liblua5.1-0 libm17n-0 libmediawiki-api-perl
  libmodule-implementation-perl libmodule-runtime-perl libotf0
  libpackage-deprecationmanager-perl libpackage-stash-perl
  libparams-classify-perl libparams-util-perl libparams-validate-perl
  libserf-1-1 libsub-install-perl libsvn-perl libsvn1 libterm-readkey-perl
  libtry-tiny-perl libyaml-perl m17n-db tla
0 upgraded, 60 newly installed, 0 to remove and 1 not upgraded.
Need to get 37.7 MB of archives.
After this operation, 190 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://archive.ubuntu.com/ubuntu xenial/main amd64 libapr1 amd64 1.5.2-3 [86.0 kB]
Get:2 http://archive.ubuntu.com/ubuntu xenial/main amd64 libaprutil1 amd64 1.5.4-1build1 [77.1 kB]
Get:3 http://archive.ubuntu.com/ubuntu xenial/main amd64 libaprutil1-dbd-sqlite3 amd64 1.5.4-1build1 [10.6 kB]
Get:4 http://archive.ubuntu.com/ubuntu xenial/main amd64 libaprutil1-ldap amd64 1.5.4-1build1 [8,720 B]
Get:5 http://archive.ubuntu.com/ubuntu xenial/main amd64 liblua5.1-0 amd64 5.1.5-8ubuntu1 [102 kB]
Get:6 http://archive.ubuntu.com/ubuntu xenial/main amd64 apache2-bin amd64 2.4.18-2ubuntu3 [918 kB]
Get:7 http://archive.ubuntu.com/ubuntu xenial/main amd64 apache2-utils amd64 2.4.18-2ubuntu3 [81.1 kB]
Get:8 http://archive.ubuntu.com/ubuntu xenial/main amd64 apache2-data all 2.4.18-2ubuntu3 [162 kB]
Get:9 http://archive.ubuntu.com/ubuntu xenial/main amd64 apache2 amd64 2.4.18-2ubuntu3 [86.6 kB]
Get:10 http://archive.ubuntu.com/ubuntu xenial/main amd64 liberror-perl all 0.17-1.2 [19.6 kB]
Get:11 http://archive.ubuntu.com/ubuntu xenial/main amd64 git-man all 1:2.7.4-0ubuntu1 [735 kB]
Get:12 http://archive.ubuntu.com/ubuntu xenial/main amd64 git amd64 1:2.7.4-0ubuntu1 [3,006 kB]
Get:13 http://archive.ubuntu.com/ubuntu xenial/universe amd64 gitweb all 1:2.7.4-0ubuntu1 [10.6 kB]
Get:14 http://archive.ubuntu.com/ubuntu xenial/main amd64 libserf-1-1 amd64 1.3.8-1 [45.5 kB]
Get:15 http://archive.ubuntu.com/ubuntu xenial/main amd64 cvs amd64 2:1.12.13+real-15 [2,332 kB]
Get:16 http://archive.ubuntu.com/ubuntu xenial/universe amd64 cvsps amd64 2.1-6ubuntu1 [40.9 kB]
Get:17 http://archive.ubuntu.com/ubuntu xenial/main amd64 emacs24-common-non-dfsg all 24.4+1-2 [4,545 kB]
Get:18 http://archive.ubuntu.com/ubuntu xenial/main amd64 emacs24-common all 24.5+1-6ubuntu1 [12.7 MB]
Get:19 http://archive.ubuntu.com/ubuntu xenial/main amd64 emacs24-bin-common amd64 24.5+1-6ubuntu1 [107 kB]
Get:20 http://archive.ubuntu.com/ubuntu xenial/main amd64 libotf0 amd64 0.9.13-3 [44.6 kB]
Get:21 http://archive.ubuntu.com/ubuntu xenial/main amd64 m17n-db all 1.7.0-2 [1,239 kB]
Get:22 http://archive.ubuntu.com/ubuntu xenial/main amd64 libm17n-0 amd64 1.7.0-3 [242 kB]
[... 340 lines cut ...]
Setting up libdatetime-format-iso8601-perl (0.08-2) ...
Setting up git-mediawiki (1:2.7.4-0ubuntu1) ...
Setting up libsvn1:amd64 (1.9.3-2ubuntu1) ...
Setting up libsvn-perl (1.9.3-2ubuntu1) ...
Setting up libyaml-perl (1.15-1) ...
Setting up libterm-readkey-perl (2.33-1build1) ...
Setting up git-svn (1:2.7.4-0ubuntu1) ...
Setting up git-email (1:2.7.4-0ubuntu1) ...
Setting up git-gui (1:2.7.4-0ubuntu1) ...
Setting up gitk (1:2.7.4-0ubuntu1) ...
Setting up git-all (1:2.7.4-0ubuntu1) ...
[... cut ...]
E: Could not open lock file /var/lib/apt/lists/lock - open (13: Permission denied)
E: Unable to lock directory /var/lib/apt/lists/
W: Problem unlinking the file /var/cache/apt/pkgcache.bin - RemoveCaches (13: Permission denied)
W: Problem unlinking the file /var/cache/apt/srcpkgcache.bin - RemoveCaches (13: Permission denied)
mint@mint ~ $ apt-get install build-essential git mtools libssl-dev
E: Could not open lock file /var/lib/dpkg/lock - open (13: Permission denied)
E: Unable to lock the administration directory (/var/lib/dpkg/), are you root?

naivePC, 2018-02-06:
if I proceed ignoring above I got error message while making the boot file

int@mint ~/thinkpad-ec $ make patched.x230.img
./scripts/generate_deps Descriptions.txt >.d/generated.deps
Downloading x230 BIOS 2.66 (G2ETA6WW) EC 1.14 (G2HT35WW)
wget -O g2uj23us.iso.orig https://download.lenovo.com/pccbbs/mobiles/g2uj23us.iso
--2018-02-06 08:13:58--  https://download.lenovo.com/pccbbs/mobiles/g2uj23us.iso
Resolving download.lenovo.com (download.lenovo.com)... 23.35.220.108
Connecting to download.lenovo.com (download.lenovo.com)|23.35.220.108|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 33927168 (32M) [application/octet-stream]
Saving to: ‘g2uj23us.iso.orig’

g2uj23us.iso.orig   100%[===================>]  32.36M   532KB/s    in 53s     

2018-02-06 08:14:51 (621 KB/s) - ‘g2uj23us.iso.orig’ saved [33927168/33927168]

scripts/checksum --rm_on_fail g2uj23us.iso.orig g2uj23us.iso
touch g2uj23us.iso.orig
./scripts/ISO_copyFL2 from_iso g2uj23us.iso.orig x230.G2HT35WW.s01D3000.FL2.orig 01D3000.FL2
git submodule update --init --remote
Submodule 'mec-tools' (https://github.com/eigenmatt/mec-tools) registered for path 'mec-tools'
Cloning into 'mec-tools'...
remote: Counting objects: 29, done.
remote: Total 29 (delta 0), reused 0 (delta 0), pack-reused 29
Unpacking objects: 100% (29/29), done.
Checking connectivity... done.
Submodule path 'mec-tools': checked out '07a1b1407326488376afed6c596d37c53d7432b6'
git submodule update
make -C mec-tools
make[1]: Entering directory '/home/mint/thinkpad-ec/mec-tools'
cc -Wall -O2 mec_csum_boot.c -o mec_csum_boot 
mec_csum_boot.c:5:19: fatal error: stdio.h: No such file or directory
compilation terminated.
Makefile:8: recipe for target 'mec_csum_boot' failed
make[1]: *** [mec_csum_boot] Error 1
make[1]: Leaving directory '/home/mint/thinkpad-ec/mec-tools'
Makefile:275: recipe for target 'mec-tools/mec_encrypt' failed
make: *** [mec-tools/mec_encrypt] Error 2

naivePC, 2018-02-06:
tried again this update was successful but 

mint@mint ~ $ sudo apt-get update
Ign:1 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial InRelease
Ign:2 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial Release
Hit:3 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib amd64 Packages
Ign:4 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib i386 Packages
Ign:5 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib all Packages
Ign:6 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib Translation-en_US
Ign:7 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib Translation-en
Hit:8 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main amd64 Packages
Ign:9 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main i386 Packages
Ign:3 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib amd64 Packages
Ign:10 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main all Packages
Ign:8 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main amd64 Packages
Ign:11 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main Translation-en_US
Ign:12 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main Translation-en
Hit:13 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free amd64 Packages
Ign:14 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free i386 Packages
Ign:13 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free amd64 Packages
Ign:15 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free all Packages
Ign:16 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free Translation-en_US
Ign:17 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free Translation-en
Ign:4 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib i386 Packages
Ign:5 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib all Packages
Ign:6 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib Translation-en_US
Ign:7 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib Translation-en
Ign:9 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main i386 Packages
Hit:3 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib amd64 Packages
Ign:10 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main all Packages
Ign:3 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib amd64 Packages
Hit:8 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main amd64 Packages
Ign:11 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main Translation-en_US
Ign:8 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main amd64 Packages
Ign:12 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main Translation-en
Ign:14 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free i386 Packages
Hit:13 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free amd64 Packages
Ign:15 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free all Packages
Ign:13 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free amd64 Packages
Ign:16 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free Translation-en_US
Ign:17 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free Translation-en
Ign:4 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib i386 Packages
Ign:5 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib all Packages
Ign:6 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib Translation-en_US
Ign:7 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib Translation-en
Ign:9 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main i386 Packages
Ign:10 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main all Packages
Hit:3 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib amd64 Packages
Ign:11 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main Translation-en_US
Ign:3 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib amd64 Packages
Hit:8 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main amd64 Packages
Ign:12 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main Translation-en
Ign:8 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/main amd64 Packages
Ign:14 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free i386 Packages
Ign:15 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free all Packages
Hit:13 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free amd64 Packages
Ign:16 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free Translation-en_US
Ign:13 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free amd64 Packages
Ign:17 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/non-free Translation-en
Ign:4 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib i386 Packages
Ign:5 cdrom://Linux Mint 18.3 _Sylvia_ - Release amd64 20171124 xenial/contrib all Packages
[... cut ...]
Fetched 6,398 B in 0s (11.7 kB/s)   
(Reading database ... 205650 files and directories currently installed.)
Preparing to unpack .../git-all_1%3a2.7.4-0ubuntu1.3_all.deb ...
Unpacking git-all (1:2.7.4-0ubuntu1.3) over (1:2.7.4-0ubuntu1) ...
Setting up git-all (1:2.7.4-0ubuntu1.3) ...
mint@mint ~ $  apt-get install build-essential git mtools libssl-dev
E: Could not open lock file /var/lib/dpkg/lock - open (13: Permission denied)
E: Unable to lock the administration directory (/var/lib/dpkg/), are you root?
mint@mint ~ $

rad4day, 2018-02-06:
```
mint@mint ~ $ apt-get install build-essential git mtools libssl-dev
E: Could not open lock file /var/lib/dpkg/lock - open (13: Permission denied)
E: Unable to lock the administration directory (/var/lib/dpkg/), are you root?
```

It's literally asking you if you executed it as superuser(root), which you didn't as you omitted the `sudo` in front of the command

hamishcoleman, 2018-02-06:
@naivePC you should never assume that carrying on regardless after an error will end with a working result.

First, you need to read the error messages - in this case, it is telling you that you need root permissions to run the apt-get command to install packages - and as @rad4day points out, you need to prefix the ```apt-get``` command with ```sudo```

naivePC, 2018-02-06:
Thank you Hamishcoleman and rad4day. Finally, I completed the task here is the description
Your build has completed with the following details:

Buildinfo: v1-205-g00321d-dirty (20180206) patched.x230.img
Built: c49c8d9b4b6fd14621f3e59cea04f113f9cee86e  x230.G2HT35WW.s01D3000.FL2

Patches applied:
x230.G2HT35WW.img.d/006_battery_validate.patch

Description: x230 BIOS 2.66 (G2ETA6WW) EC 1.14 (G2HT35WW)
./scripts/geteltorito -o patched.x230.img.tmp patched.x230.iso
Booting catalog starts at sector: 20 
Manufacturer of CD: NERO BURNING ROM
Image architecture: x86
Boot media type is: harddisk
El Torito image starts at sector 27 and has 65536 sector(s) of 512 Bytes

Image has been written to file "patched.x230.img.tmp".
./scripts/fix_mbr patched.x230.img.tmp
mv patched.x230.img.tmp patched.x230.img

Your build has completed with the following details:

Buildinfo: v1-205-g00321d-dirty (20180206) patched.x230.img
Built: c49c8d9b4b6fd14621f3e59cea04f113f9cee86e  x230.G2HT35WW.s01D3000.FL2

Patches applied:
x230.G2HT35WW.img.d/006_battery_validate.patch

Description: x230 BIOS 2.66 (G2ETA6WW) EC 1.14 (G2HT35WW)

naivePC, 2018-02-06:
Now my worry is that patch applied says "x230 BIOS 2.66 (G2ETA6WW) EC 1.14 (G2HT35WW)" but my BIOS is 2.70. Does this matter? 
Before continuing for flashing BIOS I would like to confirm this.

hamishcoleman, 2018-02-07:
Fantastic, thats a completed build - and I can confirm that you get the same sha1sum as I do.

Dont worry about the bios version number - that is not flashed to the laptop, only the EC is flashed to the laptop.  The 2.70 bios uses the same (1.14) EC version as the 2.66 bios.

Make sure you read all the info at the end of the README file with our collected notes on what we have discovered to help complete the flashing process.

naivePC, 2018-02-07:
Now I want to flash the BIOS, in the README file you have mentioned that "The firmware flash process generally requires you to have a charged   battery plugged in to the laptop before it will complete.  It may be   possible to bypass the requirement for a charged battery if you unplug   the battery completely". This means I need to take out the original battery from the laptop if I am using AC adapter right? 
I do not want to use the battery because it is almost dead it can hold maximum up to 10-15 mins only, is that sufficient if I use battery only for flashing?

rad4day, 2018-02-07:
I don't know how it's with the X230 but atleast the older Thinkpads like the T430s won't even start the flashing process unless your charger is plugged in.

naivePC, 2018-02-08:
Successfully patch done and its working. Thank you

rad4day, 2018-04-11:
@hamishcoleman I guess this Issue can be closed now :)