hamishcoleman/thinkpad-ec #67: T430 EC flash, but without using Linux?

phandojo, 2018-02-08:
Hi,

I have no experience in Linux.

After following this guide: http://www.instructables.com/id/ThinkPad-T430T430sX230-Classic-Keyboard-Mod/

I was able to successfully (physically) fit a 7-row to the T430.

But the issue is that I have 0 experience on UNIX... Only windows...

I see there's option to use .iso to patch the EC, can I burn them to a USB using Rufus and do it that way? Are there specific options in Rufus I must use to make this successful?

Thanks in advance!

diggit, 2018-02-09:
I'd try .img file and [Etcher](https://etcher.io/), but no warranty, not tested...

phandojo, 2018-02-13:
@diggit 

I managed to install ubuntu on virtualbox, will this be ok?

MooThing, 2018-02-13:
Virtualbox will do just fine for your needs, I believe you can virtually connect your USB drive to the virtual machine so that it will interact with it directly which will let you perform the dd command at the very end of the Readme without having to actually boot into Linux (I use VMware but I believe they both have this feature).

phandojo, 2018-02-13:
I cloned thinkpad-ec but I'm stumped at this step

Say I have a T430, and I want to flash the EC.

I use command "make patched.t430.img" 

The config also says it's possible to disable battery validation, so 
make patch_enable_battery clean
and
make patch_enable_keyboard clean

What would be the correct syntax to add the additional commands into "make"?

hamishcoleman, 2018-02-13:
If you want to enable patching the battery validation, and enable patching the keyboard layout for your t430:

```
make patch_enable_battery clean
make patch_enable_keyboard clean
make patched.t430.img
```

phandojo, 2018-02-13:
@hamishcoleman

Would it work if I just renamed the 006_battery_validate.patch.OFF file to 006_battery_validate.patch then use "make patched.t430.img"?

hamishcoleman, 2018-02-13:
yes, that would work - in fact, that is exactly what the the "patch_enable_battery" step does!

phandojo, 2018-02-13:
Ok thanks for the advice.

I will flash as soon as I get a working bios battery.

Do you need any info on the T430 keyboard installation? @hamishcoleman

hamishcoleman, 2018-02-13:
If you find that someone somewhere has documented the t430 install process wrong, do come back with details.  At the least, I can ensure the documentation here is accurate!

phandojo, 2018-02-13:
@hamishcoleman

Well, not necessarily wrong. But missing some details.

I will make a guide after I'm able to power the T430 on.

hamishcoleman, 2018-02-13:
Sounds good, thanks!

hwm1, 2018-02-13:
Consider selling the T430 and buying aT420.  I have both and they are practically equivalent except for the battery check and the keyboard.  If you need usb3 in the 420 just get an express card for it.  The esata connector on the 420 is handy.  I did the battery check mod on my 430 and it worked great,  I actually prefer the feel of the 430 keyboard.

hwm1, 2018-02-14:
You would break even or make money on the trade.   If there's a perceivable difference in speed it's minor.   The  expresscard usb3 ion my 420 works fine.  With ubuntu it was plug and play in fact,  Pretty amazing.

phandojo, 2018-02-14:
@hwm1

Yeah, nah.

I like the T430 as it is now.

phandojo, 2018-02-15:
@hamishcoleman 

T430 with official bios 2.74 work with the mod?

phandojo, 2018-02-15:
@hamishcoleman my bad, I think I did something wrong when "make" the img to usb, it's flashed now.

Going to install and OS and see how it goes

mikbober, 2018-03-18:
Half a day trying to get a 55+ battery to work. everything worked out.
but, unfortunately, the keyboard functions changed. how to get the working keyboard back to working t430 (fn + hot keys)

hamishcoleman, 2018-03-20:
Hi @mikbober , can you create a new issue for your problem, thanks.

mikbober, 2018-03-20:
@hamishcoleman 
So, I reflashed 2.75 BIOS with next:
make patch_enable_battery clean
make patch_disable_keyboard clean
make patched.t430.img

Hotkeys in original t430 keyboard working good.
BIOS didn't change. It's still 2.75.
Good. Thanks

hamishcoleman, 2018-04-09:
@JCBuck , yes - it is a simple matter of writing code.  Patches welcome!

closed 2018-07-03 by hamishcoleman