hamishcoleman/thinkpad-ec #68: X230 - Flashed EC firmware causes constant rapid left clicking, disappears with default firmware

MooThing, 2018-02-10:
I don't have much else to say about this issue, it occurs regardless of operating system and keyboard used (I have several to test with from a couple of broken T410 models) but ONLY when the modified EC firmware is installed. 

The classic keyboard works with all applicable hotkeys (volume, mute, media, brightness and function keys), but the left mouse button is being clicked constantly and rapidly. I have the trackpad disabled in the BIOS/UEFI which shouldn't make it a possible cause.

Please let me know if any additional information is required.

hamishcoleman, 2018-02-13:
There is nothing in the patch that touches any mouse functions, so I dont know what you are experiencing here.  If you can past the build details shown at the end of your build, I can confirm that nothing looks corrupted.

 It sounds more like the kind of issue that could happen with a hardware fault in just one keyboard (eg, if you didnt perform the isolation steps and then were unlucky with the resulting short circuit)

Actually, after re-reading your description, I'm pretty sure that the T410 keyboards are not compatible (even if they have a matching connector) - so that might be where the issue is.  Do you have any xx20 keyboards to test with?

kaggok, 2018-07-20:
Hi,
I have a classic 7 row keyboard taken out of a T410 that I want to use. Before I start, what is the difference and not compatible?

Regards.

closed 2019-11-29 by hamishcoleman