hamishcoleman/thinkpad-ec #69: No battery warning but no charging when both keyboard and battery mod are applied to an X230/

t184256, 2018-02-28:
The discovery story is here https://www.reddit.com/r/thinkpad/comments/80vp6c/help_x230_refuses_to_charge_perfectly_fine

On my X230 the mods work either-or. Keyboard patch works. Battery patch works: no warning, aftermarket battery charges. The superposition has no warning, but no charging.

hamishcoleman, 2018-02-28:
What you describe sounds quite strange.  The two patches dont actually interact with each other at all - so it they both work for you in isolation, then they should work when used together.

Do you have the detail notes from all three builds?  (The output from the build starting with "Your build has completed with the following details:")

t184256, 2018-02-28:
Yeah, I was recently [told that these should work together](https://www.reddit.com/r/thinkpad/comments/80vp6c/help_x230_refuses_to_charge_perfectly_fine/duym0ih/). Unfortunately, I didn't keep the logs, only the scripts I've built them with in the latest iteration:

```
#!/bin/bash -uex
mkdir -p images
rm -rf thinkpad-ec
git clone https://github.com/hamishcoleman/thinkpad-ec
(
cd thinkpad-ec
make really_clean patch_enable_keyboard patch_enable_battery patched.x230.img
cp patched.x230.img ../images/both.img
)
```

```
#!/bin/bash -uex
mkdir -p images
rm -rf thinkpad-ec
git clone https://github.com/hamishcoleman/thinkpad-ec
(
cd thinkpad-ec
make really_clean patch_disable_keyboard patch_enable_battery patched.x230.img
cp patched.x230.img ../images/batt.img
)
```

```
#!/bin/bash -uex
mkdir -p images
rm -rf thinkpad-ec
git clone https://github.com/hamishcoleman/thinkpad-ec
(
cd thinkpad-ec
make really_clean patch_enable_keyboard patch_disable_battery patched.x230.img
cp patched.x230.img ../images/kbd.img
)
```

Note that the hash of the resulting files was *not* reproducible from run to run.

hamishcoleman, 2018-02-28:
if the hash is not reproducible, then you sound like you have a bug with your build environment - the hash was definitely designed to be stable across rebuilds with the same patches enabled.

Without the logs, I cannot attempt to reproduce your build and look for anything strange

hamishcoleman, 2018-02-28:
actually, did you keep the built images?  If you mount the filesystem within the image, then you can extract the report.txt - which contains the build information.

t184256, 2018-02-28:
I think I have reproduced the non-charging issue under a battery-patch-only-EC, so my whole report may be a heap of misinformation. Sorry about that. Feel free to close that issue.

I think I'd 1) try to attain reproducible builds by using some other environments 2) try to match a hash for a known-good 3) reflash and reflash and reflash again.

Could you give me a hash for any one of the files obtainable with my scripts on your machine? That'd be greatly appreciated.

t184256, 2018-02-28:
Example output, if it still matters: 

```
Buildinfo: v1-211-gcd1d0a-dirty (20180301) really_clean patch_enable_keyboard patch_enable_battery patched.x230.img
Built: d1ee650c9312c91f124b85c423b7eecf0af654ef  x230.G2HT35WW.s01D3000.FL2

Patches applied:
x230.G2HT35WW.img.d/001_keysym.patch x230.G2HT35WW.img.d/002_dead_keys.patch
x230.G2HT35WW.img.d/003_keysym_replacements.patch
x230.G2HT35WW.img.d/004_fn_keys.patch x230.G2HT35WW.img.d/005_fn_key_swap.patch
x230.G2HT35WW.img.d/006_battery_validate.patch

Description: x230 BIOS 2.66 (G2ETA6WW) EC 1.14 (G2HT35WW)
```

t184256, 2018-03-01:
OK, update.

1) I thought that the .img files should be repeatable. They somehow aren't. But that does not matter, as

2) .FL2 files are repeatable and their sha1sums are googleable, so I built them correctly. But that does not matter, as

3) I've done more testing and now I can reproduce the issue with high probability (charging does not work, three green blinks, two orange ones, one green and then nothing) under both (kbd+battery) and (battery only) combinations. BUT reseating the battery about 10 times helps, until a reboot or hibernation happens.

While it's repeatable on two aftermarket batteries, I conclude that is not directly caused by the patches and is more likely some extremely strange battery wonkiness. Sorry for the noise.

closed 2018-03-01 by t184256

hamishcoleman, 2018-03-02:
@t184256 Thanks for the sample output - I was able to reproduce your build and got the same hash.

Yes, you are right - unfortunately, there are two types of files with the extension ".img", and I had forgotten that.  The usb stick filesystem files (which are only called ".img" because somebodys windows software refused to acknowledge them unless that was their name) are not reproducable.  There are two file timestamps and one date string inside the usb filesystems which will change from build to build.  I will look into removing those in the future - there is no need for most of that change