hamishcoleman/thinkpad-ec #73: T430s - Make failed

rolandch, 2018-04-14:
Getting error with "make patched.t430s.img"

Here is the output:

[https://i.imgur.com/eKsZ4kq.png](url)

Any solution?

I am using T430s and the bios is 2.70.

Thanks.

hamishcoleman, 2018-04-14:
The actual error line is the one that says "mdir: not found" - which means that It looks like you have not installed the prerequisite packages (mdir is part of the mtools package)

Did you run step 1 from the instructions?  Was that successful?

(Also, you are much better off cutting and pasting the text logs instead of using an image - it makes it easier for others to search and reference parts of it)

rolandch, 2018-04-14:
Yes, I did step 1, there was no error.

I re-installing mtools individually, I can compile now.

Thanks.

closed 2018-04-14 by rolandch

hamishcoleman, 2018-04-14:
That is strange - the package should have installed the same individually or in a group.
Oh well :-(

Thanks for confirming that it is all working for you!