hamishcoleman/thinkpad-ec #80: T430 3rd party battery patch - not working with original Lenovo 27++ battery slice

phandojo, 2018-06-10:
Hi,

As the title says, I have a T430 with the keyboard EC flash and battery EC flash that is supposed to enable compatibility with 3rd party generic batteries.

However, I seem to not be able to use a genuine Lenovo branded slice battery (27++), this is the slice battery meant for the T420 and other x20 series laptops.

Can the mod be updated to allow x30 laptop to work with x20 batteries and slice batteries (original) as well as allow 3rd party non genuine batteries to work?

Lenovo disabled the x30 laptops from charging and using the x20 batteries because they do not have the authentication chip inside.

Thanks.

hamishcoleman, 2018-07-01:
Unfortunately, I dont have a battery slice to test with, so I will not be able to do much.

You should be able to use the same techniques that zmatt originally used to find a second location in the EC firmware that is making these battery validation checks (all documented in his blog posts starting at http://zmatt.net/unlocking-my-lenovo-laptop-part-1/)

farazfazli, 2018-11-24:
@phandojo It works, be sure you are connected to a charger. I just tested it. There are 2 screens shown, one where it has the progress bar then it reboots and has another screen which says "Flashing EC". If you don't see the second screen, it didn't flash.

tabroughton, 2019-03-29:
> @phandojo It works, be sure you are connected to a charger. I just tested it. There are 2 screens shown, one where it has the progress bar then it reboots and has another screen which says "Flashing EC". If you don't see the second screen, it didn't flash.

Are you using a 27++ battery slice in a T430?  I've got this installed and I definitely flashed the EC but the slice battery still isn't charging.

tabroughton, 2019-03-30:
I got my EC to allow my battery to update.  I almost began to follow zmatt.net blog post to investigate though it would have been well over my head.  

I had to update the thinkpad-ec to grab the latest bios iso from lenovo for the T430 as I'd updated to this a few weeks ago.

Glad to confirm battery slice now charging.

closed 2019-11-29 by hamishcoleman