hamishcoleman/thinkpad-ec #81: this iso not supported  g1uj42us

sxp88, 2018-06-26:
Hello possible support t430 2.75 (G1ETB5WW) g1uj42us.iso 

thanks

hamishcoleman, 2018-07-01:
I've just had a look at the firmware download page for the t430 and can see that they did not change the EC version for any of the recent releases, so the current build will work fine.

Was there a reason why you were looking for 2.75 and not the most recent 2.77?

I have just updated the repository to use the latest 2.77 BIOS download, if that helps.

sxp88, 2018-07-02:
Hello 

 

last time i try ubuntu make patched bios 
script always download older version and not working //
now readed directly bios and moded advanced options unlocked and few other things 
and battery patch now only missing keyboard patch i dont know how to possible patch actual modded bios 

and my patched bios version 2,75 

best regards Zoltan

sxp88, 2018-07-02:
yes now downloading actual version L 
few week ago only very old version …

closed 2019-11-29 by hamishcoleman