hamishcoleman/thinkpad-ec #83: Does any of the patches (for battery validation removal only) work on x131e?

biofaust, 2018-07-11:


hamishcoleman, 2018-07-12:
Sorry, no.

I've just had a quick look at the x131e BIOS and that hardware is based on a different style of bios update files, so the existing infrastructure would need improvements to support handling these bios updates.  It is also an AMD cpu based laptop, so it is unlikely to have a EC firmware that is compatible with the current patches.

biofaust, 2018-07-12:
My x131e is Intel CPU based (i3). But indeed I guess it is significantly
different from all other ECs still.

hamishcoleman, 2018-07-15:
Ah, sorry - I must have misread some marketing about which CPU was in these systems.  Either way - when I downloaded the firmware updates, they didnt match any of the existing known patterns and I could not find the EC firmware embedded in the update to check if it was similar enough

inbx, 2018-08-29:
So, is there any way to remove batteries whitelist on x131e?
Original bios update attached. TU
[g8uj31us.zip](https://github.com/hamishcoleman/thinkpad-ec/files/2333073/g8uj31us.zip)

Kharabet, 2019-02-17:
Hi, @hamishcoleman
Is there any updates on x131e patch?

hamishcoleman, 2019-02-17:
Sorry @Kharabet , the firmware for the x131e is structured differently than the systems which we are able to remove the battery check

closed 2019-11-29 by hamishcoleman

kocoman1, 2023-09-27:
can anyone update the descrptions.txt to say where the fl2 files for the ec file is in what iso for x131e? thx

leecher1337, 2025-04-25:
Patch now available:
https://github.com/leecher1337/thinkpad-ec/tree/x131e