hamishcoleman/thinkpad-ec #86: Does Thinkpad T430s or x230 need to be reflashed for white after Lenovo bios update

rsramkis, 2018-08-07:
Hi,

My LENOVO ThinkPad T430s had the whitelist patch installed when I had the "G7ETB1WW (2.71 )" release. Recently I did the security bios update "G7ETB2WW (2.72 )"

**Question:**

Do I need to reflash this computer with the whitelist patch? Thank you

**CHANGES IN THIS RELEASE**
  Version 2.72

[Important updates]
- Security fix addresses LEN-22133 Speculative Execution Side Channel Variants 4
  and 3a (CVE-2018-3639, CVE-2018-3640). Refer to Lenovo's Security Advisory page
  for additional information. (https://support.lenovo.com/product_security/home)

rad4day, 2018-08-07:
Depends if the patch included a new EC image. If it did, you'll most likely get notified on the next boot after that your battery is not supported and will not be charged. If not, you're probably going to be fine.

rsramkis, 2018-08-07:
Thanks for the feedback. If I have to reflash, are there any issues using he same IMG as before?

hamishcoleman, 2018-08-07:
@rsramkis - I just wanted to confirm what you mean when you say "whitelist patch" - are you referring to the EC patch to allow third party batteries? Or perhaps a wifi whitelist patch?

I've just looked at the t430s 2.72 release and it does not include any EC version change, so any EC patches should not be overwritten by applying the BIOS update.

Reusing the old EC IMG patch should work fine if there are no EC version changes, however for best results, it is always recommended to update to the latest thinkpad-ec github release and rebuild the patch - and also to confirm that the version numbers used in the patch match the version numbers currently in your system.

rsramkis, 2018-08-07:
@hamishcoleman - When I applied the bios patch to the T430s and x230 I used the following options from the config.txt file:

    make patch_enable_battery clean     # Uses the battery validate patch
    make patch_disable_keyboard clean # Turns off the keyboard patches

The t430s img.d had the following on:
006_battery_validate.patch

The x230 img.d had the following on:
006_battery_validate.patch

I did not see any mention for wifi whitelist or any other whitelist in the config.txt.
Do I need to do anything to have the wifi whitelist activated?

hamishcoleman, 2018-08-07:
Sorry to confuse you - it was just your use of the word "whitelist" - I wanted to be sure we were talking about the same things!  I've worked it out now and you are all good - just ignore the first paragraph of my previous reply.

(There are other projects which try to change the wifi whitelist, but there is nothing in this project that can adjust that)

rsramkis, 2018-08-07:
@hamishcoleman Thanks Hamish. Good opportunity for me to just learn more about your teams efforts while running a budget thinkpad. I appreciate all the support.i believe all questions answered. We can close this one. ☺

closed 2018-08-07 by rsramkis