hamishcoleman/thinkpad-ec #87: [Question] Coreboot and thinkpad-ec

dannycolin, 2018-08-10:
Sorry if I ask a dumb question :)

So I was wondering if it is possible to have Coreboot and a modified EC. If yes, how should I proceed?

Thanks

hamishcoleman, 2018-08-11:
This is not a dumb question!

It is possible to have both a modifed EC and Coreboot, but you need to apply the EC patch before installing Coreboot.

The low-level method of programming the EC with a new firmware is not known, so there is no way for Coreboot to apply a modified EC.

dannycolin, 2018-08-11:
Eh thanks a lot.

closed 2018-08-11 by dannycolin

simontretter, 2018-09-17:
` → Chipset ───────────────────────────────────────────────────────────────────────────────────── 
  ┌──────────────────────────────────── Add EC firmware ─────────────────────────────────────┐   
  │ CONFIG_HAVE_EC_BIN:                                                                      │   
  │                                                                                          │   
  │ The embedded controller needs a firmware file.                                           │   
  │                                                                                          │   
  │ Select this if you are going to use the PCH integrated controller                        │   
  │ and have the EC firmware. EC firmware will be added to final image                       │   
  │ through ifdtool.                                                                         │   
  │                                                                                          │   
  │ Symbol: HAVE_EC_BIN [=n]                                                                 │   
  │ Type  : boolean                                                                          │   
  │ Prompt: Add EC firmware                                                                  │   
  │   Location:                                                                              │   
  │     -> Chipset                                                                           │   
  │       -> Add Intel descriptor.bin file (HAVE_IFD_BIN [=y])                               │   
  │   Defined at src/southbridge/intel/common/firmware/Kconfig:123                           │   
  │   Depends on: HAVE_INTEL_FIRMWARE [=y] && HAVE_IFD_BIN [=y]                              │     `

hamishcoleman, 2018-09-17:
@simontretter , that is interesting.

However, I dont think that this is a complete solution.  From a quick look at the source (I looked in src/southbridge/intel/common/firmware/Makefile.inc and util/ifdtool), building coreboot with HAVE_HC_BIN means that an additional flash partition is filled with the provided EC binary image - however, none of the x*30 thinkpad systems have a EC controller that loads its program from the SPI flash.

As I understand it, the vendor flashing sequence is to:
1. use dosflash.exe to fill this SPI-flash partition with the EC bin
2. set a flag somewhere in the BIOS boot sequence
3. Reboot
4. while in early boot, notice the flag (and continue to normal boot if no flag)
5. handshake with the EC, before it has been fully setup to manage the hardware
6. upload the new EC firmware to the EC chip, using an unknown protocol
7. Probably tell the EC to reboot
8. Reboot the laptop
9. Probably do some confirmation that the flash completed OK
10. clear the flag

Unfortunately, the HAVE_HC_BIN looks like it helps to deal just with step 1 of that sequence - but we still would need to know how do to steps 5, 6 and 7