hamishcoleman/thinkpad-ec #90: X230 BIOS ver 2.68 - how to make patch if I don't want to update?

paralemptor, 2018-09-07:
Let's say I do not want to update my BIOS to the newest. 2.73 version. How do I make the patch so it will be compatible with 2.68? When I ran the patch to generate the .img file it automatically downloaded the 2.73 BIOS version...

hamishcoleman, 2018-09-07:
A very reasonable question.

There are two answers:

1. Looking at the BIOS and EC versions list, I can see that the EC version has not changed between 2.68 and 2.73, so the .img you prepared should be fine to use anyway.

2. Slightly more work, but more obviously correct - edit the Descriptions.txt to switch which version is used to build the .img - In this case, the line starting with "x230.G2HT35WW.s01D3000.FL2" is the definition for the EC version you need, and both the dep and depi parameters need to be updated.  Currently they both point at the latest ISO filename ("g2uj29us.iso"), so change them to point at the 2.68 ISO filename ("g2uj25us.iso")

paralemptor, 2018-09-07:
Thank you for your fast response! I will probably try the already prepared .img file as soon as my keyboard arrives and report back if anything breaks ;)

paralemptor, 2018-09-10:
Just did the method number two, just in case. Everything went well. Thank you very much!

closed 2018-09-10 by paralemptor