hamishcoleman/thinkpad-ec #91: Patching EC firmware in modern ThinkPad notebooks

kitsunyan, 2018-09-08:
Hi!

Recently I was obsessed with the idea of installing T25's keyboard into T480. I already tried to physically connect the keyboard to my friend's T480. All "normal" keys worked as expected, but Fn keys didn't: their behavior reminds me how X220's keyboard works in X230 without the patch.

Then I decided to dive into your work on EC research. I tried to find keyboard table in T470's EC firmware (T470 and T25 share the same firmware) and I found… 4 of them, which contradicts to the information provided here https://github.com/hamishcoleman/thinkpad-ec/blob/150fef43c4064754cc46ee522b13efec0b230fef/docs/table_matrix.txt#L111 (only 2 tables were found). I tried to check N1QHT32W firmware (there is a typo in your docs: you call N1QHT32W as N1QHT57W, probably being confused by BIOS version N1QET57W) and found that there were 2 tables, but that was before T25 support was added. That means there is a branching to chose which table to use depending on the model name or something (the absence of reverse engineering skills doesn't allow me to give a certain answer), from which I can assume that the same branching exists for Fn keys.

Funny enough that all 4 tables are the same. And they're the same on T480, which explains why T25's keyboard works this good in T480. But I still haven't found the information about how Fn keys handled in modern ThinkPads (I'm working on it…). Perhaps, I can found Fn keys table for T470 and T480, compare them and then apply the table for T25…

But it all doesn't make any sense since I don't have an idea how can I apply any patches. As you described here https://github.com/hamishcoleman/thinkpad-ec/blob/150fef43c4064754cc46ee522b13efec0b230fef/docs/firmware_layout.txt#L63-L65 there is a signature in the bottom. I tried to compare N1QET50W and N1QET51W and the most major difference between them is the footer. https://gist.github.com/kitsunyan/67e4f89233d44036af25dea187d0f358

What does this signature mean? If it uses public-key cryptography, does that mean there it no way we can modify EC firmware on new ThinkPad notebooks? Do you know any researches regarding ECs and their verification processes in modern ThinkPads?

nayjerenko, 2019-01-22:
Interesting idea! Did you make any progress? Does the keyboard fit mechanically?

kitsunyan, 2019-01-23:
Yes, [I finished my mod](https://old.reddit.com/r/thinkpad/comments/a9xhx3), but without patching the EC. I partially fixed the keyboard layout using [kernel module](https://github.com/kitsunyan/fpkbd), although without proper EC patch my solution is OS-dependent and some keys are lost.

comradeKGBagent, 2020-04-24:
Hello, I was wondering if any progress was made. 

Im currently working on a mod myself, and have made a small bit of progress with regards to flashing the EC itself, but not with finding the correct firmware. 

From my understanding, there is not a way to flash the chip through software, so it has to be done with a hardware flashed (like SVOD3) that supports MEC1653. The flash is done through the JTAG 2 port on the motherboard. 

It may be that the keyboard matrix for T25 isnt present on T470 (which would be why you did not find it) meaning that a new keyboard matrix would have to be written from scratch. 

I know leokim on r/thinkpad was able to do this, however he shares very little info in his post. Ive contacted him, but am yet to hear back. I hope to have a solution soon.