hamishcoleman/thinkpad-ec #97: T430 keyboard

bigkappa, 2018-11-11:
Hi. 
I have recently bought a used T430 and would like to mod the Keyboard (and probably the battery later). My current UEFI BIOS Version is G1ET94WW (2.54), E.C G1HT33WW (1.11). I've seen that Lenovo released recently a new Bios Update (2.78, Oct. 1st 2018). I would like to know if i can update to that version and then apply the mod in your repository.

Thanks

hamishcoleman, 2018-11-11:
I've just had a look at the new 2.78 Bios and since it is using the same EC version as the previous Bios release, it was very simple to update this repo to use it.

You should be good to update your t430 Bios to the latest version and then apply patches from this repo.

bigkappa, 2018-11-11:
Awesome. Many thanks for your fast reply. For what i understood, with this patches we are able to able/disable those features whenever we prefer. Correct? It's a mod on the actual BIOS or custom feature on it and not actual a different BIOS. I will always be able to turn off the patch or restore the default BIOS from warehouse or upgrade with the lates version (in that case I could lose the improvements by your patches).

closed 2019-08-06 by hamishcoleman