hamishcoleman/thinkpad-ec #98: Trackpoint drifting after install

WudUpA, 2018-11-13:
This issue is not a result of the 7-row keyboard install process.  It has been an issue on some thinkpads for a long time.  However, after i installed the keyboard and flashed the EC, Everything worked fine.  After about 10 - 60 minutes, the trackpoint would start drifting away.  The problem is easily solved by changing the sensitivity of the trackpoint.  After a reboot, the sensitivity returns to the default setting. My question is...Is the default setting in the EC firmware? If so, how do i change the default setting and reflash?

hamishcoleman, 2018-11-14:
Unfortunately, while the trackpoint is physically connected to the embedded controller, nobody has done any research into looking at the firmware related to this.

The kind of research needed would be getting to know the raw PS/2 protocol used to talk to the trackpoint and then looking at the firmware data tables for anything that looks like initialisation data related to that protocol.

Normally, the trackpoint should only drift occasionally - and then only for a short time while it recalibrates.  I understand that this issue is something that a lot other people have experienced, but it does also sound like a faulty (or marginally working) trackpoint module..

WudUpA, 2018-11-15:
Thanks for taking the time to answer my questions.  I know your a busy guy and I appreciate it!

closed 2019-08-06 by hamishcoleman