hamishcoleman/thinkpad-ec #101: x230 BIOS Update 2.73 safe to apply patch?

Zzausch, 2018-11-22:
Hey @hamishcoleman I am quite a noob but managed to flash the keyboard mod on my x230 about 1 year ago (I was on BIOS 2.66 I think), now i would like to patch and flash the new BIOS 2.73 (with the spectre/meltdown fixes: https://pcsupport.lenovo.com/de/de/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x230/downloads/ds029187) - is the keyboard and battery patch safe to apply to this newest bios? Just want to be super-safe an not brick my device with a incorrectly patchd BIOS. Also I have no linux machine running at the moment, is there an easy way to do this on windows?

Thx!

naggie, 2019-03-05:
How about 2.74?

naggie, 2019-03-06:
For anyone else wondering -- it appears to be supported as a known version as it's listed in Descriptions.txt. I've just built the patched ISO and will report back when I've flashed it.

naggie, 2019-03-06:
It worked! Though, not from USB. I burned the image to CD for the flash, and now my battery is working. Thanks @hamishcoleman for this project!

hamishcoleman, 2019-03-06:
hey @naggie.  As you found, yes the 2.74 is the current version in the repository.

Just wondering if there was a reason you used a CD instead of a USB stick?  Did you try using the USB and find that it did not work?

naggie, 2019-03-06:
Yes exactly, I writing to a USB stick with dd. Selecting it from the boot menu (F12) just went back to the same boot menu.

tabroughton, 2019-03-30:
> Yes exactly, I tried writing to a USB stick with dd. Selecting it from the boot menu (F12) just went back to the same boot menu.

I was having similar issue, I was deploying image to a partition I created on the usb (/dev/sdb1) once I deploed to /dev/sdb (without partition) it worked.

closed 2019-08-06 by hamishcoleman