hamishcoleman/thinkpad-ec #103: Issue Compiling patched.t430.img

homeInOuray, 2018-11-28:
I have been trying to reapply the patch suite to my T430, as I have done in the past, but keep running into this error on any machine I try to compile on:

scripts/checksum --mv_on_fail g1uj45us.iso.orig g1uj45us.iso
touch g1uj45us.iso.orig
./scripts/ISO_copyFL2 from_iso g1uj45us.iso.orig t430.G1HT35WW.s01D2000.FL2.orig 01D2000.FL2
mcopy: File "::/FLASH/$01D2000.FL2" not found
make: *** [.d/generated.deps:17: t430.G1HT35WW.s01D2000.FL2.orig] Error 1

I can't seem to make sense as to what the issue is, so i figured I'd post it here. Also, huge thanks to hamishcoleman for the incredible work on this project. Keep it up!

hamishcoleman, 2018-11-29:
hrm, looks like it could be a minor incompatibility with one of the tools used.

What linux distribution are you running?

Are you able to send the output from running running:

`sh -x ./scripts/ISO_copyFL2 from_iso g1uj45us.iso.orig t430.G1HT35WW.s01D2000.FL2.orig 01D2000.FL2`

As that might help debug things

homeInOuray, 2018-11-29:
Here's my output:
`+ FAT_OFFSET=71680
+ DIR=from_iso
+ case "$DIR" in
+ shift
+ ISO=g1uj45us.iso.orig
+ '[' '!' -e g1uj45us.iso.orig ']'
+ shift
+ which mdir
+ '[' from_iso = ls ']'
+ FILENAME=t430.G1HT35WW.s01D2000.FL2.orig
+ '[' -z t430.G1HT35WW.s01D2000.FL2.orig ']'
+ shift
+ INTPATTERN=01D2000.FL2
+ '[' -z 01D2000.FL2 ']'
++ mdir -i g1uj45us.iso.orig@@71680 -/ -b
++ grep 01D2000.FL2
+ MATCH='::/FLASH/$01D2000.FL2'
+ '[' -z '::/FLASH/$01D2000.FL2' ']'
++ echo '::/FLASH/$01D2000.FL2'
++ wc -w
+ '[' 1 -ne 1 ']'
+ from_iso
+ mcopy -n -i g1uj45us.iso.orig@@71680 '::/FLASH/$01D2000.FL2' t430.G1HT35WW.s01D2000.FL2.orig
mcopy: File "::/FLASH/$01D2000.FL2" not found
`

I'm on Arch Linux with everything as up-to-date as they can be.

hamishcoleman, 2018-11-30:
hrm, that is a strange one.

It looks like the line that is failing is `mdir -i g1uj45us.iso.orig@@71680 -/ -b |grep 01D2000.FL2`

- I get the output "`::/FLASH/G1ETB8WW/$01D2000.FL2`"
- you are getting "`::/FLASH/$01D2000.FL2`"

Can you paste the output of both  `mtools --version` and `mdir -i g1uj45us.iso.orig@@71680 -/ -b`

homeInOuray, 2018-11-30:
Strange indeed. If you're on Sid ten you should be pretty much just as up-to-date.

Here's my output

mtools --version:

`mtools (GNU mtools) 4.0.21
configured with the following options: enable-xdf disable-vold disable-new-vold disable-debug `enable-raw-term`

mdir -i g1uj45us.iso.orig@@71680 -/ -b:

`::/COMMAND.COM
::/WINA20.386
::/CONFIG.SYS
::/AUTOEXEC.BAT
::/HIMEM.SYS
::/RAMDRIVE.SYS
::/FLASH/
::/EFI/
::/System Volume Information/_restore{5AAE7142-4278-4615-A947-75249691AF85}/
::/System Volume Information/_restore{2CE34FFD-D3EF-4519-A9E8-7515A6204392}/
::/System Volume Information/IndexerVolumeGuid
::/System Volume Information/WPSettings.dat
::/System Volume Information/LightningSand.CFD
::/FLASH/UTILINFO.EXE
::/FLASH/G1ETB8WW/
::/FLASH/206A7.PAT
::/FLASH/306A2.PAT
::/FLASH/306A4.PAT
::/FLASH/306A5.PAT
::/FLASH/306A6.PAT
::/FLASH/306A8.PAT
::/FLASH/306A9.PAT
::/FLASH/BCP.EVS
::/FLASH/CHKLOGOD.EXE
::/FLASH/COMMAND.COM
::/FLASH/DOSFLASH.EXE
::/FLASH/FINDRAMD.EXE
::/FLASH/FLASH2.EXE
::/FLASH/lcreflsh.bat
::/FLASH/README.TXT
::/FLASH/SHELLFLASH.EFI
::/FLASH/TPCHKS.EXE
::/FLASH/UPDTFLSH.EXE
::/FLASH/UPDTMN.EXE
::/FLASH/USERINT.EXE
::/FLASH/$01D2000.FL1
::/FLASH/$01D2000.FL2
::/FLASH/206a7.hsh
::/FLASH/306a4.hsh
::/FLASH/306a5.hsh
::/FLASH/306a6.hsh
::/FLASH/306a8.hsh
::/FLASH/306a9.hsh
::/EFI/BOOT/
::/EFI/BootX64.efi`

Also, thanks for looking into this!

homeInOuray, 2018-11-30:
Interesting. Is there a way for me to work around this temporarily? I just need to create a drive with all the patches enabled.

hamishcoleman, 2018-11-30:
You can try manually running `mcopy -n -i g1uj45us.iso.orig@@71680 '::/FLASH/G1ETB8WW/$01D2000.FL2' t430.G1HT35WW.s01D2000.FL2.orig`

and then restarting the build.

However, it runs this command again later in the build, so you will need to run a different manual version of the same command later.

Another option is to use any other released version of mtools from the last 10 years - even the previous version 4.0.20 works fine - so if the arch repos still have an earlier package version, you could just downgrade to it

homeInOuray, 2018-11-30:
You're a legend. Built and ready to go. Thank you so much!

hamishcoleman, 2018-12-03:
I have submitted a bug report to the mtools mailing list: https://lists.gnu.org/archive/html/info-mtools/2018-12/msg00001.html

closed 2019-08-06 by hamishcoleman