hamishcoleman/thinkpad-ec #105: x230.G2HT35WW.img.d/007_capslock_ctrl_swap.patch Does not apply

ode79, 2018-11-29:
```
$ make patched.x230.iso
./scripts/generate_deps Descriptions.txt >.d/generated.deps
./scripts/ISO_copyFL2 from_iso g2uj29us.iso.orig x230.G2HT35WW.s01D3000.FL2.orig 01D3000.FL2
./scripts/FL2_copyIMG from_fl2 x230.G2HT35WW.s01D3000.FL2.orig x230.G2HT35WW.img.enc.tmp
IMG at offset 0x500000 size 0x30000 (FL2::prefix_ff x230.G2HT35WW.s01D3000.FL2.orig)
mec-tools/mec_encrypt -d x230.G2HT35WW.img.enc.tmp x230.G2HT35WW.img.orig.tmp
mec-tools/mec_csum_flasher -c x230.G2HT35WW.img.orig.tmp
64c4 64c4 OK
mec-tools/mec_csum_boot -c x230.G2HT35WW.img.orig.tmp
b4b73a78 b4b73a78 OK
f02cdcf8 f02cdcf8 OK
3ff064dd 3ff064dd OK
40da02e2 40da02e2 OK
rm x230.G2HT35WW.img.enc.tmp
mv x230.G2HT35WW.img.orig.tmp x230.G2HT35WW.img.orig
cp --reflink=auto x230.G2HT35WW.img.orig x230.G2HT35WW.img
./scripts/hexpatch.pl --rm_on_fail --report x230.G2HT35WW.img.report x230.G2HT35WW.img x230.G2HT35WW.img.d/*.patch
Attempting to patch x230.G2HT35WW.img
Address 137616 mismatched data
The binaryfile does not match the context bytes from x230.G2HT35WW.img.d/007_capslock_ctrl_swap.patch
make: *** [Makefile:186: x230.G2HT35WW.img] Error 1
```

Bios information from dmidecode.
```
Handle 0x000C, DMI type 0, 24 bytes
BIOS Information
	Vendor: LENOVO
	Version: G2ETB3WW (2.73 )
	Release Date: 06/19/2018
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 12288 kB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		3.5"/720 kB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 2.73
	Firmware Revision: 1.14
```

hamishcoleman, 2018-11-30:
From the look of your output there, you are applying only the 007_capslock_ctrl_swap.patch and none of the other patches.  This patch is applied on top of the other keyboard patches and is dependant on them.

Additionally, swapping your capslock and ctrl keys is something that your Operating System can do, so patching the EC firmware is quite an overkill for this task.

hamishcoleman, 2019-02-14:
@ode79, did you try again with a different set of patches enabled?

ode79, 2019-02-16:
I didn't understand that this project was only for the old 7 row keyboards when I opened the issue.

The issue isn't relevant to me currently.

Thanks anyway.

closed 2019-02-16 by ode79