hamishcoleman/thinkpad-ec #107: Caps lock LED

stefanct, 2019-01-01:
Currently the caps lock LED in the caps lock key of pre-30-series keyboards does not work properly in modified 30-series systems. On pre-30 systems the `LEDCPSLOCK` signal is controlled by the EC and connected to pin 21 of the keyboard connector via a current limit resistor (560 ohm for the t420s to 820 ohm for the t410s) resulting in approximately 3-5mA. On 30-series systems that pin 21 is connected to signal `KBD_BL_DTCT` (probably keyboard backlight discrete-time-to-continuous-time) which is an *input* to the EC (I am not 100% sure about its purpose but apparently some feedback from the keyboard). It is attached to ball/pin `D2` of the MEC1619 EC and has a 100kOhm pullup. That pin can be internally multiplexed to the VCI block (VBAT-powered Control Interface) of the EC or act as ordinary GPIO pin. Without looking at the software I assume it is not used connected to the VCI but used as GPIO.
The maximum output current of `D2` is 4mA.¹

So from the electrical side of things it looks to me that we could simply switch the direction of the pin and use it to drive the caps lock LED if we could alter the firmware accordingly. I have not tried to asses that possibility from the software side but I hope somebody here can comment on the feasibility of that w/o too much effort. Is there any major obstacle to implement this functionality?

¹ Since there is no public data sheet of the MEC1619 my statements above rely on its familiarity to the MEC1618 which is documented [here](http://ww1.microchip.com/downloads/en/DeviceDoc/00002339A.pdf).

TheShadowOfChernobyl, 2019-01-03:
KBD_BL_DTCT probably stands for "Keyboard backlight detect" and could be an identification pin to tell the EC whether the installed keyboard is a back-lit version or not.

Tafnab, 2019-06-04:
I have a question about the capslock too. But I did something a bit different. I have a W530. I swapped the circuitry from the x20 series keyboard with that from the x30 series. The caps lock is stuck on. I now have a x20 series illuminated keyboard (doesn't work that well). Is the stuck caps lock light a thing because of the circuitry I changed, or is it a thing that needs to have the patched bios to fix?

frederiiiic, 2020-04-25:
Hi there,

I've just monted a X220 keyboard on my x230 laptop, with the BIOS update everything is working fine. Big thanks to the work done here. But i'm very sad to see that the caps lock led is not working, glad to see that it's interesting not only me!

Hope someone will find a solution.