hamishcoleman/thinkpad-ec #109: [question] t580 unlocked bios

dannycolin, 2019-01-13:
Hi @hamishcoleman,

A couple of month ago, someone on reddit bought a t580 and it turns out it was a laptop from someone from Lenovo. The BIOS and all the advanced options were unlocked. The buyer uploaded a BIOS dump on the internet and I managed to get a copy.

I was wondering would that BIOS be of any help to you?

hamishcoleman, 2019-01-22:
That is an interesting development.  I'm not sure that it would be much help to me, but we should make sure that it does not disappear from the internet..

token0, 2019-04-14:
@dannycolin I would like to occasionally find a chip dump on some fs site like Mega.nz. Strictly in personal educational purpose. I want to mod my ThinkPad (E, not T or W), but can't yet find a way to **add** advanced menu tab (though I found a right module and it's GUID), and not just **swap** first "Main" tab with "Advanced". So I need some hook to make this mod. By the way, which UEFI does it have, Insyde or Phoenix? Maybe you can share a screenshot of menu?