hamishcoleman/thinkpad-ec #111: T530 question from a noob.

routomatic, 2019-01-20:
Good day sir. 

Today I just ran into a problem after buying an after market battery for my t530.

Unfortunately, it does not charge.

I am a total noob when it comes to linux and programming in general. 

I just want to know if I can safely follow your method on my system.

Here are some info.

UEFI BIOS VERSION: G4ET92WW (2.52)
EMBEDDED CONTROLLER VERSION G4HT39WW (1.31)

Also, can I use a portable version of linux.

Sorry for all these questions.

I will really appreciate if you can help me.

hamishcoleman, 2019-01-22:
Your system has an older UEFI BIOS version, but it is still using a supported EC version.  So, yes, you can use this method on your system.  It would probably be better if you upgrade your BIOS to the latest version before starting.

If you portable linux can install the needed compiler and tools, then there should be no issue with using that.

routomatic, 2019-01-24:
Got it. thank you.

closed 2019-01-24 by routomatic