hamishcoleman/thinkpad-ec #113: Is keyboard patch working for others?

gerb58, 2019-01-25:
I have been having trouble for the last couple of weeks getting the keyboard patch to work on the W530.

One month ago I was successful enabling vintage keyboard, a couple weeks ago I was successful in disabling the vintage keyboard.

I have a different W530 now.

But now, following the same procedures, I cannot boot from a created USB. I simply refreshes the boot menu.

Have tried the usual bios settings changes. Have tried multiple USB devices. 

I am stuck. I don't know if this issue is on my end or not.

gabidanpremier, 2019-01-27:
Hi guys ! I'm really impressed by what you've accomplished so far. I have the exact same issue stated here and I also started looking at patching my W530 bios yesterday! It's funny because ... timing!!

I want the battery patch for my W530. I run Ubuntu 18.04 TLS on Hyper-V. I don't have access to neither a DVD writer nor a USB dongle directly from this VM, so I have to use Windows 10 in the way of making a bootable DVD or USB key.

Here's what I've done so far (grouped to make sense):
1- The ISO from Lenovo (g5uj36us.iso) can be burned (using the Windows tool) and works (can be booted by the computer).
2- The ISO g5uj36us.iso.orig has the same MD5 as the Lenovo one, so that's alright.
3- The ISO patched.w530.iso (burned the same way) cannot be booted by the computer.
4- **The ISO patched.w530.iso doesn't have the same MD5 from one "make" to the other** (3 out of 3). This doesn't seem related to my local file transfer as the Lenovo ISO files checks out and was transferred the same way.
5- The patched.w530.img doesn't boot either. I'm using Rufus 3.4 to make a bootable USB key from an IMG file.
6- I tried using the BIOS update utility. I installed the Lenovo BIOS utility and copied all the files from the USB key created using the patched IMG file into the utility folder. I then ran the BIOS update utility hoping it would push all the patched files. I ended up with an updated BIOS, but without the patch. I think that because there was no EC firmware version change, it didn't update it with the patch.

The next things I could try is using an older version of the code, as of November 1st for example, before the 2.73 BIOS update. Appart from that, I'm pretty much of ideas.

Any hint?

gabidanpremier, 2019-01-27:
I found a workaround. gerb58, you may be interested.

As I proposed an hour ago, I got back to an older version of the BIOS (an older branch). I'm still a newbie with GitHub, so here's how to do it:

Between step 3 and 4:
`git reset --hard 80a5dd882a`

It built the 2.72 BIOS, which I then burned to a DVD. I booted with it, it patched the EC firmware.

Now to have the 2.73 BIOS, I guess I could simply use the Lenovo BIOS update utility which shouldn't update the EC firmware.

gerb58, 2019-01-27:
It works! Thank you!

lolENGEN, 2019-02-07:
Same issue with the T530 as well

hamishcoleman, 2019-02-14:
@lolENGEN , are you able to confirm which version of this repository you had issues with?  Also, if I make the same workaround as for the w530, are you able to test that the resulting image does boot?

hamishcoleman, 2019-02-15:
Hey, @gabidanpremier and @gerb58, I have looked at the ISO files downloaded from Lenovo, and I can see what looks like an error that was introduced in their g5uj36us.iso - if I update the code to try patching that, are one of you able to build a bootable image and test if it can boot?

lolENGEN, 2019-02-16:
Sorry for the late response, I have done just as @gabidanpremier has done but with BIOS 2.74 already installed and only flashed the EC. I'd be happy to test for you, just let me know.

lolENGEN, 2019-02-16:
To me it looks like the T530 BIOS haven't been updated too often, after this https://github.com/hamishcoleman/thinkpad-ec/commit/08ad6971849ec5bc47648117a5f0ca14b2853fce
I couldn't boot into the flasher.

forgive me if it seems like I don't know this site very well, I actively made this account just for this so feel free to correct me.

hamishcoleman, 2019-02-16:
Cool, thanks @lolENGEN - I can see that there is a 1-byte difference in the boot code in the image from Lenovo between the known working and known broken versions.

I have added some code to detect this and try to repair it in the latest commit - if you could try booting that, that would be great!

lolENGEN, 2019-02-23:
Just created and flashed the latest build and it boots perfectly, I used the ubuntu build from the windows store so I didn't have to flash linux again and it worked first try. Nice work!

vanceb, 2019-08-06:
I am also having this issue with an X230.  Build process seems fine, but the resulting image (.img for USB Stick) and ISO (CD) won't boot.  I am only attempting the battery patch upgrade not the keyboard...

I have successfully upgraded to the latest BIOS 2.76 and EC 1.14 using the Lenovo files without modification before attempting to flash the patched BIOS.

When building this source for the patched version, it does give the following "warning" during build:

`Found corrupted bootcode in ISO from Lenovo - attempting fix`

But I am not sure this is relevant.

Build Process when making the .img file:

~~~ Shell
make patched.x230.img
Generated dependancies from descriptions
./scripts/ISO_copyFL2 from_iso g2uj31us.iso.orig x230.G2HT35WW.s01D3000.FL2.orig 01D3000.FL2
./scripts/FL2_copyIMG from_fl2 x230.G2HT35WW.s01D3000.FL2.orig x230.G2HT35WW.img.enc.tmp
IMG at offset 0x500000 size 0x30000 (FL2::prefix_ff x230.G2HT35WW.s01D3000.FL2.orig)
mec-tools/mec_encrypt -d x230.G2HT35WW.img.enc.tmp x230.G2HT35WW.img.orig.tmp
mec-tools/mec_csum_flasher -c x230.G2HT35WW.img.orig.tmp >/dev/null
mec-tools/mec_csum_boot -c x230.G2HT35WW.img.orig.tmp >/dev/null
./scripts/hexpatch.pl --rm_on_fail --report x230.G2HT35WW.img.report x230.G2HT35WW.img x230.G2HT35WW.img.d/*.patch
Attempting to patch x230.G2HT35WW.img
Applying x230.G2HT35WW.img.d/006_battery_validate.patch -7294,7 +7294,7
./scripts/xx30.encrypt x230.G2HT35WW.img x230.G2HT35WW.img.enc.tmp
+ cd mec-tools
+ ./mec_repack ../x230.G2HT35WW.img ../x230.G2HT35WW.img.enc.tmp
building ../x230.G2HT35WW.img.enc.tmp from ../x230.G2HT35WW.img
b4b73a78 b4b73a78 OK
f02cdcf8 f02cdcf8 OK
3ff064dd 203e469e FIXED
40da02e2 40da02e2 OK
64c4 0962 FIXED
98e8 7041 FIXED

reverifying
7041 7041 OK
0962 0962 OK
b4b73a78 b4b73a78 OK
f02cdcf8 f02cdcf8 OK
203e469e 203e469e OK
40da02e2 40da02e2 OK
./scripts/FL2_copyIMG to_fl2 x230.G2HT35WW.s01D3000.FL2.tmp x230.G2HT35WW.img.enc.tmp
IMG at offset 0x500000 size 0x30000 (FL2::prefix_ff x230.G2HT35WW.s01D3000.FL2.tmp)
./scripts/ISO_copyFL2 to_iso g2uj31us.iso.tmp x230.G2HT35WW.s01D3000.FL2.tmp 01D3000.FL2
mcopy -t -m -o -i g2uj31us.iso.tmp@@71680 g2uj31us.iso.report.tmp ::report.txt
mcopy -t -m -o -i g2uj31us.iso.tmp@@71680 g2uj31us.iso.bat.tmp ::AUTOEXEC.BAT
mdel -i g2uj31us.iso.tmp@@71680 ::EFI/Boot/BootX64.efi
cp g2uj31us.iso patched.x230.iso
cp g2uj31us.iso.report patched.x230.iso.report

Your build has completed with the following details:

Buildinfo: v1-307-ge5b632-dirty (20190805) patched.x230.img
Built: 611c88f62add31aa4cdbedeec50e9583af509cbe  x230.G2HT35WW.s01D3000.FL2

Patches applied:
x230.G2HT35WW.img.d/006_battery_validate.patch

Description: x230 BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW)
./scripts/geteltorito -o patched.x230.img.tmp patched.x230.iso
Booting catalog starts at sector: 20 
Manufacturer of CD: NERO BURNING ROM
Image architecture: x86
Boot media type is: harddisk
El Torito image starts at sector 27 and has 65536 sector(s) of 512 Bytes

Image has been written to file "patched.x230.img.tmp".
./scripts/fix_mbr patched.x230.img.tmp
Found corrupted bootcode in ISO from Lenovo - attempting fix

Your build has completed with the following details:

Buildinfo: v1-307-ge5b632-dirty (20190805) patched.x230.img
Built: 611c88f62add31aa4cdbedeec50e9583af509cbe  x230.G2HT35WW.s01D3000.FL2

Patches applied:
x230.G2HT35WW.img.d/006_battery_validate.patch

Description: x230 BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW)
~~~

vanceb, 2019-08-06:
Managed to get a version working...

Reverted back to an earlier commit which specifically mentions X230.

`git reset --hard cd447e6`

Then followed the build process to create the .img file.  Burned the ISO - which wouldn't boot (still)!

Tried a flash drive image (on a new flash drive - the old drive may also have been an issue) which worked!

So I am not really sure whether the flash drive was a problem earlier, but the latest ISO didn't work either.  Although I didn't specifically build the .iso, I used the one which was created as part of the build for the .img file.

hamishcoleman, 2019-08-06:
@vanceb The warning you saw is not a build error, it is just a comment on the data that came from Lenovo - I have changed the wording output to try and clarify that.

The version mismatch concern is only relevant when there is a different EC version (both the 2.76 and 2.75 BIOS use the same 1.14 EC version) - so that's not an issue.  This also does not affect the bootablility of the image.

There are basically no changes between the e5b632 version that did not work for you and the cd447e6 version that did work - No process or code changes and the only change that touches the x230 was to mention (but not use) the 2.76 BIOS.

I dont know that many people try burning an ISO - so it is possible that that has not worked for a while (there is also no difference between asking for the ISO to be built and just using the one that is built automatically)

In summary, you appear to have done nothing wrong - so I would be looking at the hardware.

If you wanted to participate in some testing, we can try and find out what does and does not work.

vanceb, 2019-08-13:
I have reverted to the master branch commit, built and created a USB stick that boots and flashes the EC.  I had to remove the HDD to get the bios to look at the USB stick (since changed the boot order in the BIOS), but this was not the issue before.  

Previously I had managed to get the boot device menu up (F12), and was selecting the USB Stick to boot from, but after 5 seconds or so was presented again with the boot device menu as if there was nothing to boot from that device.  

I will try the ISO when I get chance to see if I can eliminate that as a problem.

vanceb, 2019-08-13:
Just tried the ISO and it doesn't want to boot. It attempts a boot, fails (silently) then presents me with the BIOS boot menu.  Selecting the CD as the boot device from the menu attempts a boot then re-presents the boot menu.

aquohn, 2021-05-03:
Hi, I'm experiencing the same symptoms as @vanceb: both ISO and CD fail at the boot selector screen. Setting the CD as the first option in the boot order appears to cause the system to hang at the Lenovo splash screen. I'm keen to help carry out some tests to determine what is going on!

qharvest, 2022-07-24:
> Hi, I'm experiencing the same symptoms as @vanceb: both ISO and CD fail at the boot selector screen. Setting the CD as the first option in the boot order appears to cause the system to hang at the Lenovo splash screen. I'm keen to help carry out some tests to determine what is going on!
> 
> EDIT: I tried rebooting with the USB at the top of the boot order list; it presented a black screen stating "Disk error; press any key to reboot". After that it booted into my SSD, and attempts after that to boot into the USB drive failed. For reference, I'm using a T530, with BIOS 2.65 (G4ETA5WW) and EC 1.13 (G4HT39WW).

Hi. I have the exact same issue right now. I've successfully updated my T430 BIOS to the latest patchable BIOS version using the image from Lenovo's website. However, I can't boot on the patched version, whenever I select my USB stick it just doesn't do anything else than refreshing the Boot Menu.

Have you found any workaround? That would really help me a lot. Thank you.