hamishcoleman/thinkpad-ec #114: T430 "non-genuine battery" message dissappeared but battery is still not charging

mrquantos, 2019-02-17:
Hi, Thank you for creating this project. You are doing great.

I have after market T430 70++ battery. I have updated bios to latest version 2.78 and applied battery patch to my T430. EC flashed successfully and the lenovo "non-genuine battery" message has disappeared from boot screen but battery is still not charging. Can you please help me on this issue?

If it helps, here is "tlp" output:

```
--- TLP 0.8 --------------------------------------------

+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (kernel module 'tp_smapi' not installed)
tpacpi-bat = active

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SANYO
/sys/class/power_supply/BAT0/model_name                     = 45N1015
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  71280 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  72710 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =   5100 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown (threshold effective)

tpacpi-bat.BAT0.startThreshold                              =     96 [%]
tpacpi-bat.BAT0.stopThreshold                               =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0
```
If you need anything else, just let me know.
Thank you!

mrquantos, 2019-02-21:
@hamishcoleman please could you suggest, is there anything else I could try to make it work? 

Thank you!

lolENGEN, 2019-02-23:
I have a T430 as well so I'll try to duplicate this issue and try to fix it as well.

lolENGEN, 2019-02-24:
Just tested, it worked fine.

Make sure you have it plugged in with a currently working battery to activate the EC flash (after it loads the BIOS on the first boot it should reboot to a screen that says something like , EC flashing DO NOT POWER OFF!) unless you see that, it didn't work.

mrquantos, 2019-02-24:
@lolENGEN Thank you for testing it out. 

I have flashed EC patch with my genuine battery as you suggested and it did show that EC flashing message. Now, I don't get lenovo's "unauthorized" battery message anymore which means EC flashed successfully and patch is working but my after market battery is still not charging which is odd. 

Would sniffing I2C communication, at the moment of battery installation, help? I have logic analyzer which works with [Sigrok Pulseview](https://sigrok.org/wiki/PulseView)

lolENGEN, 2019-02-24:
Seems like it might be some bad settings in tlp.
As I'm not running Linux currently I am unfortunately not much help but I did find a nice walkthrough https://linrunner.de/en/tlp/docs/tlp-faq.html#battery from https://askubuntu.com/questions/1050488/tlp-will-not-let-my-battery-charge that had a similar issue. Hope it helps.

mrquantos, 2019-02-26:
@lolENGEN Thank you for taking time to help me out. I have tried settings for tlp as you suggested but it did not work. May be there is something wrong with my battery, may be it is faulty. I will purchase another battery. Anyway, thank you!

closed 2019-02-26 by mrquantos